mne-tools/mne-python #12654: Export EDF using `mne.export.export_raw`: Signal range error when original data range does not include zero AND data is not equal-length

qian-chu, 2024-06-10:
### Description of the problem

In `mne.export.export_raw`, if the user specifies `physical_range="auto"`, the physical range is inferred from the data. Meanwhile, if the data record is not an exact multiple of the data record length (1 sec by default), edfio appends additional zeroes to the end.
If the original data range does not include zero (all positive or all negative), then this creates an interesting `ValueError: Signal range [0.0, 2.0] out of physical range: [1.0, 2.0]`
Users can of course override this by manually setting the `physical_range` argument in `mne.export.export_raw`. But do you think it makes more sense to automatically detect this scenario and extend physical_range to include zero?

### Steps to reproduce

```Python
import numpy as np
from mne import create_info
from mne.io import RawArray

# 999 data points, not a multiple of 100
data = np.concatenate([np.ones(500), np.ones(499)*2]) * 1e-6
data = np.tile(data, (5, 1))
ch_names = np.arange(5).astype(str).tolist()
info = create_info(ch_names=ch_names, sfreq=100, ch_types=['eeg']*5)
raw = RawArray(data, info)

raw.export('test_range.edf', overwrite=True)
```

### Expected results

Zeroes are appended without triggering ValueError

### Actual results

`ValueError: Signal range [0.0, 2.0] out of physical range: [1.0, 2.0]`

### Additional information

Platform             Windows-10-10.0.22631-SP0
Python               3.11.5 | packaged by conda-forge | (main, Aug 27 2023, 03:23:48) [MSC v.1936 64 bit (AMD64)]
Executable           C:\Users\QianC\.conda\envs\mnedev\python.exe
CPU                  Intel64 Family 6 Model 165 Stepping 2, GenuineIntel (16 cores)
Memory               15.8 GB

Core
├☑ mne               1.8.0.dev94+g0293e7bc6 (devel, latest release is 1.7.0)
├☑ numpy             1.24.4 (MKL 2022.1-Product with 8 threads)
├☑ scipy             1.11.2
└☑ matplotlib        3.7.2 (backend=QtAgg)

Numerical (optional)
├☑ sklearn           1.3.0
├☑ numba             0.57.1
├☑ nibabel           5.1.0
├☑ nilearn           0.10.1
├☑ dipy              1.7.0
├☑ openmeeg          2.5.6
├☑ pandas            2.1.0
├☑ h5io              0.1.9
├☑ h5py              3.9.0
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.42.1 (OpenGL 4.5.0 - Build 26.20.100.7985 via Intel(R) UHD Graphics)
├☑ pyvistaqt         0.0.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.0 (PySide6=6.5.2)
├☑ ipympl            0.9.3
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.5.2
├☑ ipywidgets        8.1.0
├☑ trame_client      2.11.2
├☑ trame_server      2.11.7
├☑ trame_vtk         2.5.8
└☑ trame_vuetify     2.3.1

Ecosystem (optional)
├☑ mne-bids          0.14.dev0
├☑ mne-connectivity  0.6.0dev0
├☑ neo               0.12.0
├☑ eeglabio          0.0.2-4
├☑ edfio             0.4.2
├☑ mffpy             0.8.0
├☑ pybv              0.7.5
└☐ unavailable       mne-nirs, mne-features, mne-icalabel, mne-bids-pipeline

drammock, 2024-06-17:
> Users can of course override this by manually setting the `physical_range` argument in `mne.export.export_raw`. But do you think it makes more sense to automatically detect this scenario and extend physical_range to include zero?

seems worthwhile and probably not too difficult. Do you want to give it a shot @qian-chu ?

hofaflo, 2024-06-19:
> Should this be addressed upstream @hofaflo?

Don't think so, zero-padding is done in [`mne/export/_edf.py`](https://github.com/mne-tools/mne-python/blob/d082461bcb701fd66b186e92535fb5288f047c38/mne/export/_edf.py#L68), so the behavior of edfio is intended. The issue is that physical min and max are taken from the un-padded data [here](https://github.com/mne-tools/mne-python/blob/d082461bcb701fd66b186e92535fb5288f047c38/mne/export/_edf.py#L87-L96).

cbrnr, 2024-06-19:
Thanks!

@qian-chu @drammock I'm not sure if automatically extending the physical range to include zero is the best option. Could there be a situation where this might have unintended effects? I'm thinking of signals with large offsets, and if the physical range is automatically extended, the 16 bits of resolution will be spread over a much larger range of physical values.

Instead, could we maybe pad with the last available value instead of zero?

qian-chu, 2024-06-19:
> Thanks!
>
> @qian-chu @drammock I'm not sure if automatically extending the physical range to include zero is the best option. Could there be a situation where this might have unintended effects? I'm [...]

I think this would be the best approach! And we can maybe add a `BAD_padding` annotation to denote that the padding is not real signal. Should be easy to implement too

drammock, 2024-06-19:
> Could there be a situation where this might have unintended effects? I'm thinking of signals with large offsets, and if the physical range is automatically extended, the 16 bits of resolution will be spread over a much larger range of physical values.

That is a logical possibility but one that to me seems unlikely to occur in the wild (though maybe I'm wrong). To me the proposed alternative (to pad with a (very large) non zero value) seems very odd / likely to cause problems later.

larsoner, 2024-06-19:
> To me the proposed alternative (to pad with a (very large) non zero value) seems very odd / likely to cause problems later.

One way to think of it is like [`np.pad` modes](https://numpy.org/doc/stable/reference/generated/numpy.pad.html) and one would be `constant` (zeros) and the proposed one would be the second option `edge`. Thinking about it like a padding problem there usually isn't one right answer and each has some tradeoff. If there is a risk of messing with the resolution then it might actually be safer to stick with the edge values because we're at least guaranteed not to worsen any depth problem (whereas with zeros we could!). So I'm +0.5 for padding with the edge values. We could eventually consider making it an option but I'd expect the edge values to be safer than zeros for limited-precision storage.

Also instead of `BAD_padding` a more consistent choice I think would be `BAD_ACQ_SKIP`, which is what we use elsewhere when a file encompasses a span of time when data was not actually acquired.

cbrnr, 2024-06-19:
Like @larsoner said, there is no single best solution, but why would a zero value be different from any other value? Especially if the data is not centered anyway, I'd think that padding with the edge value is the safer choice here.

drammock, 2024-06-19:
Ok I misunderstood "last available value". I thought it was meant to be the largest representable value given the bit depth of the file format.

Padding with the temporally last value makes a lot of sense given the downside of using zero.

qian-chu, 2024-06-19:
If everyone's down for replacing zero padding with edge padding I can initiate a short PR any time :) Also I noticed there's no padding test so it would be handy to also add it

PR #12676 (2024-06-20): Change EDF zero-padding to edge-padding with annotation and test
Fixes #12654.

#### What does this implement/fix?
Changes how `mne.export.export_raw(..., fmt='edf')` handles data that cannot be segmented into equal-length data blocks. The original zero-padding is not optimal for uncentered data and may unexpectedly enlarge the physical range. Here, zero-padding is replaced with `np.pad(array, pad_width, mode='edge'`. The last available value in each channel will be used to pad the data. It also adds a "BAD_ACQ_SKIP" annotation to denote the padded data.

Corresponding tests also added.
closed 2024-06-20 by larsoner