mne-tools/mne-python #12685: Can we disable the "Back to top" button in the docs?

hoechenberger, 2024-06-26:
It's constantly obstructing part of the content I want to read. And it's super irritating this thing constantly appears and disappears, based on if and how and in which direction I'm scrolling. FYI I've not used a mouse in like 10 years, I do everything with a trackpad, because it makes pixel-precise scrolling so easy – so I scroll a lot. The "Back to top" link is super not great for me. Thoughts on this?

[image]

cbrnr, 2024-06-26:
I find this annoying too, 👍 to remove it.

drammock, 2024-06-26:
It's configurable, can be disabled in conf.py

Also on my list of things to tweak about how we use the theme. Will do when back from holidays; feel free to do it yourselves first if you don't want to wait

PR #12688 (2024-07-13): Disable the "Back to top" button in the documentation
Closes #12685
closed 2024-07-13 by cbrnr