mne-tools/mne-python #12709: Make full installation depend on PySide6 by default

hoechenberger, 2024-07-11:
Since the bug with interactive plotting when using PySidd6 has now been [fixed](https://bugreports.qt.io/plugins/servlet/mobile#issue/PYSIDE-2192), we should be able to replace the PyQt6 dependency in `mne[full]` with PySide6.

- PySide6 is the official Qt for Python binding.
- Binary wheels of PySide6 is available for more platforms, including `aarch64`.
- The conda-based installation – which is what we recommend in our docs – and our installers will use PySide6 starting with the next MNE-Python release.
- PySide6 is available from conda-forge, while PyQt6 is not (yet). Switching the dependency to to PySide6 will make for a cleaner and smoother experience in situations where a conda-forge and PyPI-based installation of MNE-Python are mixed.

The above mentioned bug fix should be included in the upcoming release 6.7.3.

Lastly, probably more controversial, but: my opinion is that once we've made the switch, we can (should) drop support for PyQt and get rid of qtpy, which just adds another layer of complexity that
won't be needed anymore.

cbrnr, 2024-07-11:
I agree :100:%! Once PySide6 is available and works, qtpy is obsolete because there is no need to support another Qt binding such as PyQt.

There are two additional points in favor of PySide6 that I'd like to add:

- PySide6 is licensed under the LGPL, whereas PyQt6 uses the GPL. It's not entirely safe to use GPL-licensed code in a BSD- or MIT-licensed project.
- PySide6 is being updated to the latest Qt versions much more rapidly and frequently (probably because it is the official binding). For example, PyQt6 has been stuck at 6.7.0 for months now, whereas PySide6 closely follows the Qt release cycle.

larsoner, 2024-07-11:
Okay with switching to a `PySide6>=6.7.3` pin once it actually lands.

I'd rather not get rid of `qtpy` -- it has very little maintenance overhead / burden for us and makes our library compatible with more Qt bindings, including PyQt5 which is still widely used on conda-forge. Maybe in a year or two once things settle it could make sense to switch to PySide6, but I don't think we should move fast on this.

cbrnr, 2024-07-11:
I think people should *not* be using an old Qt binding, and dropping qtpy would ensure that. I understand if you don't want to do it right now, but we should eventually get rid of it (a year or two seems too long, let's maybe reconsider in half a year or so).

hoechenberger, 2024-07-11:
(on a somewhat related side node I always found it extremely funny that it's the Spyder devs who came up with qtpy, yet not only do they strictly depend on PyQt5, but even a very specific version of it. I find this ironic and it kind impair my trust in the approach qtpy has taken)

wmvanvliet, 2024-07-14:
-1 from me on dropping qtpy because I would like Qt5 support a bit longer. On our workstations in our department, we don't update our entire anaconda environment that fast and the machines used by our researcher to get their daily work done are still on Qt5 (because Qt6 is not available yet through conda-forge).

cbrnr, 2024-07-14:
Qt6 is available through conda-forge via PySide6.

hoechenberger, 2024-07-14:
Yes Qt6 has been available on conda-forge for almost 2 years now. Qt6 is more than 3.5 years old.

hoechenberger, 2024-07-14:
That said I don't feel strongly about qtpy, so for now i'm rather neutral on this one. I just think it's one bit we should definitely consider getting rid of whenever we believe it's convenient!

wmvanvliet, 2024-07-14:
> Yes Qt6 has been available on conda-forge for almost 2 years now. Qt6 is more than 3.5 years old.
> 
> Honest question, if the workstations don't get updated anyway – can you not just stick with an older MNE-Python, then?

wow I feel silly. I always thought pyqt6 would be the official package, but it is pyside6 and that has been available for a long time. I never knew. The solver always refused to install pyside6 because we are also users of Spyder, which depends on pyqt5... This is an informative thread for me.

hoechenberger, 2024-07-14:
Yes Spyder is super odd, they do use qtpy but depend on PyQt5 😵‍💫
And I don't think you're the first one who believes PyQt6 is the official binding. In fact I would be the majority of Python users would answer this question incorrectly. It's such a mess on many levels.

larsoner, 2024-08-20:
FYI I just tried to use PyQt6 and then PySide6 with vscode's Python debugger and ran into https://github.com/microsoft/debugpy/issues/1488 and https://github.com/microsoft/debugpy/pull/1569. So Qt6 support is still in the works / incomplete in `VSCode` in addition to `conda` and `Spyder` apparently!

larsoner, 2024-08-20:
Looks like @wmvanvliet has been busy trying to make things better there at least :)

https://github.com/microsoft/debugpy/commit/0035e83a40ba77080f9ae8b829835b2a0c4834f6

cbrnr, 2025-06-17:
Note that meanwhile, PySide6 has resolved the long-standing issue regarding interactive plots: https://bugreports.qt.io/browse/PYSIDE-2192

So I think we could just require PySide6 >= 6.8 to be safe.

cbrnr, 2025-06-17:
I think there's a significant difference here. MNE automatically shows interactive plots, which Matplotlib doesn't. So unless we change that behavior (which I have suggested previously) to be consistent with Matplotlib, I don't think we should then also just follow what Matplotlib is doing.

cbrnr, 2025-06-17:
> I see what you're saying but I am not totally convinced by this. Yes it's a little inconsistent that we have `show=True` and matplotlib has `show=False` but even when you do `plt.show()` in matplotlib it emits a warning, so we're consistent at least in what the user sees, even though we have a different default attempted behavior. The `show` in both places basically means "try to show it if you can / are in a framework that allows it". So I don't think our `show=True` default is a convincing reason to deviate from what matplotlib does / recommends / installs.

That's not entirely correct. While the user sees a warning when doing `plt.show()`, there is absolutely nothing when doing `raw.plot(show=True, block=True)`. I think this needs to be fixed regardless of making PySide6 a core dependency or not.

Regarding your comments on use cases, to me interactive plotting is something that 95% of all users will want to do almost immediately. I do not have any numbers to back this up, but from my experience and talking to other people, I think this estimate is not far off. Currently, those users who did `pip install mne` will not see a plot, no matter what they do. They have to manually `pip install pyside6`. If you are not willing to make PySide6 a core dependency, then there should at least be some warning/error when plotting and no interactive backend is available. But again, this means that most users will have to `pip install mne pyside6` for basic functionality!

I understand your point about people wanting a minimal subset of functionality, but then I suggest that we also remove Matplotlib. Why would we want to force it upon users if it is not functional for all Linux and Windows users in the first place?

wmvanvliet, 2025-06-18:
I would expect `pip install mne` to have functional interactive plotting out of the box, while I'd expect `pip install mne[base]` to be "headless".

drammock, 2025-06-18:
> I would expect `pip install mne` to have functional interactive plotting out of the box, while I'd expect `pip install mne[base]` to be "headless".

unfortunately the `project.optional-dependencies` cannot *remove* packages declared under `dependencies`. So that's not possible.

cbrnr, 2025-06-18:
I think these are both great suggestions. Publishing two packages is certainly nice, but could be substantially more effort. Better advertising seems like the easier alternative, but I'm really OK with either solution.

Regarding the absence of any message: I thought that this was exactly the reason why we're still using `qtpy`. I remember at some point seeing a message along the lines of "No Qt bindings found, please install PySide6, PyQt6, ...". Did this stop working at some point?

cbrnr, 2025-06-19:
For me, it's very simple. Currently, a typical user will not be able to use interactive plots out of the box. Therefore, I'd try to make this easier for the user by adding PySide6 to the core dependencies. I don't know what the problem here is, people running MNE on a headless Linux server surely will not be negatively affected if this package gets installed automatically.