mne-tools/mne-python #12710: Consider using absolute imports

cbrnr, 2024-07-11:
I'd like to suggest to use absolute instead of relative imports. Absolute imports are more readable and in general provide better error messages, so PEP8 (and I 😄) recommend(s) using them (https://peps.python.org/pep-0008/#imports). Most of the work can probably be done by Ruff (https://docs.astral.sh/ruff/rules/relative-imports/).

hoechenberger, 2024-07-11:
FWIW MNE-BIDS has been using absolute imports for many years, maybe @sappelhoff would care to share the reasoning behind and experience with that?

cbrnr, 2024-07-11:
Although this is a separate issue, I'm also +💯 on switching to a src layout. I've also used both things (absolute imports and src layout) in MNELAB and SleepECG without any problems (not surprisingly, because these are the recommended approaches nowadays).

agramfort, 2024-07-11:
>
> Although this is a separate issue, I'm also +💯 on switching to a src
> layout. I've also used both things (absolute imports and src layout) in
> MNELAB and SleepECG without any problems (not surprisingly, because these
> are the recommended approaches nowadays).
>
numpy, scipy, scikit-learn, joblib don't use src layout...

no objection with absolute imports

> Message ID: ***@***.***>
>

hoechenberger, 2024-07-11:
Sorry about conflating this issue / request here, let's maybe not consider / discuss src layout for now and focus on absolute imports only :)

sappelhoff, 2024-07-11:
> FWIW MNE-BIDS has been using absolute imports for many years, maybe @sappelhoff would care to share the reasoning behind and experience with that?

I have always tried to recommend / enforce this (when I had the decision power), for the same reasons that Clemens mentions. :-) I would be very happy if we adopted this in MNE-Python, too.

hoechenberger, 2024-07-11:
> I have always tried to recommend / enforce this (when I had the decision power),

Yes, it was actually you who made me research this topic a little after you rejected my proposal to switch everything to relative imports a couple of years back or so :) this is when I figured that absolute imports are actually recommended.

larsoner, 2024-07-11:
Okay with me to migrate to absolute imports. It'll probably create a lot of PR conflicts so we should make sure we don't have any big pending PRs open when we do it (or at least be aware of what they are so we can consciously decide to accept the rebase/merge pain)

drammock, 2024-07-11:
> Okay with me to migrate to absolute imports.

no strong feeling. When doing so we should update our import/nesting tests to detect and fail if anyone puts in a relative import.