mne-tools/mne-python #12715: Lots of symbols in `mne.utils` are unknown to my static type checker, probably due to an issue with lazy loading

hoechenberger, 2024-07-15:
### Description of the problem

Lots of functions imported from `mne.utils` allegedly don't exist according to Pylance; however, they **are** there, but obviously the static type checker cannot find them. For example here's what I see for a bunch of functions imported from `utils.check`:
[image]

Red lines – Import not found

I took a look at `utils.pyi` and here, a large portion of imports is just black (i.e., not linked to the imported object because it couldn't be found):

[image]

### Steps to reproduce

```Python
See above
```

### Expected results

Static type checker should be aware of the imported objects

### Actual results

Static type checker cannot find the imported objects

### Additional information

This is with `main` and latest VS Code, VS Code Python, and Pylance extensions.

cbrnr, 2024-07-15:
This might be an issue related to your particular setup/config, because it works for me...

hoechenberger, 2024-07-15:
How did you install MNE and from what directory are you working / invoking VS Code?

cbrnr, 2024-07-15:
MNE is installed from my up-to-date local clone via `pip install -e .`, and I've set my project directory to the root of the MNE clone.

hoechenberger, 2024-07-15:
Thanks for checking this, @cbrnr!
I completely re-created my env and now it's working again, so not an MNE bug after all – great! Sorry for the noise.

closed 2024-07-15 by hoechenberger