mne-tools/mne-python #12750: Som font colors in the dark theme are not readable

hoechenberger, 2024-07-25:
### Proposed documentation enhancement

[image]

https://mne.tools/stable/auto_tutorials/preprocessing/35_artifact_correction_regression.html#references

scott-huberty, 2024-08-05:
Also, as seen in the [Eyetracking tutorial:](https://mne.tools/stable/auto_tutorials/preprocessing/90_eyetracking_data.html)

[image]

drammock, 2024-08-05:
working on this right now

hoechenberger, 2024-08-05:
> Also, as seen in the [Eyetracking tutorial:](https://mne.tools/stable/auto_tutorials/preprocessing/90_eyetracking_data.html)

Ha, I think this was one of those "we'll fix it later" bits back when we merged my revised HTML reprs. I'll take full blame for this one; however I won't have time to look into this before September or so. But happy to test on macOS if @drammock can come up with a solution before that

PR #12776 (2024-08-05): Website fixes
This PR:

- re-does the fixes to navbar icon colors from #12301 (CSS rule broke because fontawesome icons are now `<svg>` instead of `<i>` tags)
- fixes color contrast problems on sphinx-gallery page elements (closes #12750)
- fixes sidebar quicklink icon colors on dark theme
- fixes frontpage gallery text colors on dark theme

Does *not* address https://github.com/mne-tools/mne-python/issues/12750#issuecomment-2269424078 for which #12775 has been opened.
closed 2024-08-05 by larsoner