mne-tools/mne-python #12764: MNE.NIRS/nilearn.plotting Errors

pgk713, 2024-08-01:
### Description of the problem

I am new to MNE Nirs (have used MNE with EEG for quite sometime) and have been trying follow the sample code provided on the MNE Nirs website however i keep encountering the same error when I try and install the packages necessary to run the code. I have redownloaded and pip installed the newest version of MNE and i still am encountering errors.

When I try and import mne.nirs it spits out the following error:
17 import mne_nirs
     18 from mne_nirs.channels import get_long_channels, get_short_channels, picks_pair_to_idx
     19 from mne_nirs.experimental_design import make_first_level_design_matrix

File ~\anaconda3\lib\site-packages\mne_nirs\__init__.py:17, in <module>
     15 from . import signal_enhancement
     16 from . import simulation
---> 17 from . import statistics
     18 from . import utils
     19 from . import visualisation

File ~\anaconda3\lib\site-packages\mne_nirs\statistics\__init__.py:7, in <module>
      5 from ._roi import glm_region_of_interest
      6 from ._statsmodels import statsmodels_to_results
----> 7 from ._glm_level_first import (RegressionResults, ContrastResults,
      8                                run_glm, run_GLM, read_glm)

File ~\anaconda3\lib\site-packages\mne_nirs\statistics\_glm_level_first.py:12, in <module>
     10 import numpy as np
     11 import pandas as pd
---> 12 from h5io import read_hdf5, write_hdf5
     13 from numpy import array_equal, where
     15 with warnings.catch_warnings(record=True):

ModuleNotFoundError: No module named 'h5io'

Similarly, when i try and import "from nilearn.plotting import plot_design_matrix" i get the following error: ModuleNotFoundError: No module named 'nilearn'

All of the code i am using is the sample code provided on the MNE NIRS website but am still encountering these issues. If anyone has a way to fix this it would be very helpful and very much appreciated!

### Steps to reproduce

```Python
import mne

from nilearn.plotting import plot_design_matrix
import mne_nirs
from mne_nirs.channels import get_long_channels, get_short_channels, picks_pair_to_idx
from mne_nirs.experimental_design import make_first_level_design_matrix
from mne_nirs.statistics import run_glm
```

### Expected results

to load packages properly

### Actual results

error messages above

### Additional information

any help would be greatly appreciated

agramfort, 2024-08-01:
what do you mean by redownloaded and pip installed? how did you install
python on your machine? did you use the mne installer?

Alex

Message ID: ***@***.***>
>

pgk713, 2024-08-01:
I dont know what you mean about the MNE installer, is there a standalone to download MNE onto my computer? I run all my code through jupyter notebook but I redownloaded the new version of MNE into my jupyter note book using the code "!pip install MNE" in my kernal. I then used the code provided to pip install the MNE NIRS onto my jupyter notebook by running the code: "!pip install -U --no-deps git+https://github.com/mne-tools/mne-nirs.git@main" .

It says that MNE and MNE NIRS are successfully downloaded after i run the above code however when i try and run the code "import mne.nirs" to call the packages into the new script it gives me the error mentioned above about how theres No module named 'h5io'. the same thing happens when i try to import nilearn as well.

hoechenberger, 2024-08-01:
Hello, the error messages suggest you need to install h5io and nibabel.

Best wishes,
Richard

pgk713, 2024-08-01:
Here is the code that i am running to import these modules.  both give me the same error that the modules dont exist so im not sure how to get them installed?
 
from nilearn.plotting import plot_design_matrix
    
ModuleNotFoundError: No module named 'nilearn'

When I try and import mne.nirs it spits out the following error:
17 import mne_nirs
18 from mne_nirs.channels import get_long_channels, get_short_channels, picks_pair_to_idx
19 from mne_nirs.experimental_design import make_first_level_design_matrix

File ~\anaconda3\lib\site-packages\mne_nirs_init_.py:17, in
15 from . import signal_enhancement
16 from . import simulation
---> 17 from . import statistics
18 from . import utils
19 from . import visualisation

File ~\anaconda3\lib\site-packages\mne_nirs\statistics_init_.py:7, in
5 from ._roi import glm_region_of_interest
6 from ._statsmodels import statsmodels_to_results
----> 7 from ._glm_level_first import (RegressionResults, ContrastResults,
8 run_glm, run_GLM, read_glm)

File ~\anaconda3\lib\site-packages\mne_nirs\statistics_glm_level_first.py:12, in
10 import numpy as np
11 import pandas as pd
---> 12 from h5io import read_hdf5, write_hdf5
13 from numpy import array_equal, where
15 with warnings.catch_warnings(record=True):

ModuleNotFoundError: No module named 'h5io'

hoechenberger, 2024-08-01:
you can install them via pip

pgk713, 2024-08-01:
i tired to !pip install nilearn and got this error:
ERROR: Could not install packages due to an OSError: [WinError 5] Access is denied: 'C:\\Users\\pkell\\anaconda3\\Lib\\site-packages\\~umpy\\core\\_multiarray_tests.cp39-win_amd64.pyd'
Consider using the `--user` option or check the permissions.

hoechenberger, 2024-08-01:
Hello, can you please ask these questions in our user forum at https://mne.discourse.group ? this issue tracker is not really the right place for this. Thank you and good luck!

closed 2024-08-01 by hoechenberger