mne-tools/mne-python #12767: Cannot open nedf file from Neuroelectrics

JoaoAmaro2001, 2024-08-01:
### Description of the problem

When I run:
```python
p = r'C:\Users\joaop\pessoal\20240627111911_OE204007_EXP4.nedf'
mne.io.read_raw_nedf(p)
```

I get the following error:
{
	"name": "ParseError",
	"message": "no element found: line 1, column 0 (<string>)",
	"stack": "Traceback (most recent call last):

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\xml\\etree\\ElementTree.py:1713 in close
    self.parser.Parse(b\"\", True) # end of data

ExpatError: no element found: line 1, column 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\IPython\\core\\interactiveshell.py:3577 in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  Cell In[6], line 3
    mne.io.read_raw_nedf(p)

  File <decorator-gen-455>:12 in read_raw_nedf

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\mne\\io\
edf\
edf.py:227 in read_raw_nedf
    return RawNedf(filename, preload, verbose)

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\mne\\io\
edf\
edf.py:139 in __init__
    header, dt, dt_last, n_samp, n_full = _parse_nedf_header(header)

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\mne\\io\
edf\
edf.py:72 in _parse_nedf_header
    headerxml = defusedxml.ElementTree.fromstring(header[:headerend])

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\defusedxml\\common.py:127 in fromstring
    return parser.close()

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\xml\\etree\\ElementTree.py:1715 in close
    self._raiseerror(v)

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\xml\\etree\\ElementTree.py:1615 in _raiseerror
    raise err

  File <string>
ParseError: no element found: line 1, column 0
"
}

### Steps to reproduce

```Python
p = r'C:\Users\joaop\pessoal\20240627111911_OE204007_EXP4.nedf'
mne.io.read_raw_nedf(p)
```

### Expected results

Dataset

### Actual results

{
	"name": "ParseError",
	"message": "no element found: line 1, column 0 (<string>)",
	"stack": "Traceback (most recent call last):

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\xml\\etree\\ElementTree.py:1713 in close
    self.parser.Parse(b\"\", True) # end of data

ExpatError: no element found: line 1, column 0

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\IPython\\core\\interactiveshell.py:3577 in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)

  Cell In[6], line 3
    mne.io.read_raw_nedf(p)

  File <decorator-gen-455>:12 in read_raw_nedf

  File c:\\Users\\joaop\\Anaconda3\\envs\\data_science\\Lib\\site-packages\\mne\\io\
edf\
edf.py:227 in read_raw_nedf
    return RawNedf(filename, preload, verbose)

### Additional information

Platform             Windows-11-10.0.22631-SP0
Python               3.12.4 | packaged by conda-forge | (main, Jun 17 2024, 10:04:44) [MSC v.1940 64 bit (AMD64)]
Executable           c:\Users\joaop\Anaconda3\envs\data_science\python.exe
CPU                  Intel64 Family 6 Model 186 Stepping 2, GenuineIntel (20 cores)
Memory               15.7 GB

Core
├☑ mne               1.7.1 (latest release)
├☑ numpy             2.0.0 (MKL 2024.1-Product with 14 threads)
├☑ scipy             1.14.0
└☑ matplotlib        3.9.1 (backend=widget)

Numerical (optional)
├☑ sklearn           1.5.1
├☑ nibabel           5.2.1
├☑ pandas            2.2.2
└☐ unavailable       numba, nilearn, dipy, openmeeg, cupy, h5io, h5py

Visualization (optional)
├☑ vtk               9.3.1
├☑ qtpy              2.4.1 (PyQt5=5.15.8)
├☑ ipympl            0.9.4
├☑ ipywidgets        8.1.3
└☐ unavailable       pyvista, pyvistaqt, pyqtgraph, mne-qt-browser, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
├☑ mne-bids          0.15.0
└☐ unavailable       mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

JoaoAmaro2001, 2024-08-03:
Hello @larsoner, I have tried using neuroelectrics' plugin for EEGLAB and I also get an error. I have checked the .nedf file and the version is 1.4.

JoaoAmaro2001, 2024-08-03:
Hello @larsoner, I have tried using neuroelectrics' plugin for EEGLAB and I also get an error. I have checked the nedf file and the version is 1.4.

larsoner, 2024-08-03:
If other software also falls, it suggests that the file is corrupt. I'll close but let us know if you manage to read the file using something else. Maybe it didn't copy from one system / drive to another?

closed 2024-08-03 by larsoner