mne-tools/mne-python #12768: Should `fetch_fsaverage()` return a `Path`?

hoechenberger, 2024-08-02:
Currently, `mne.datasets.fetch_fsaverage()` returns a string, while all `datasets.*.data_path()` functions return a `Path`. I find this inconsistent, esp. in tutorials, where we'd want to use both, `fetch_fsaverage()` and `data_path()`.

Thoughts?

cbrnr, 2024-08-02:
I'm always for improving consistency, so if that's what other functions are doing I'm 👍. In addition, it might be worth looking a bit into the `mne.datasets` package, because to me it seems to be a bit confusing since there are many functions with possibly overlapping functionality and/or naming:

- `mne.datasets.fetch_dataset()`: Is this supposed to be the wrapper for downloading each and every dataset?
- If that's the case, why is there are separate `mne.datasets.fetch_fsaverage()`?
- What about the various `mne.datasets.XXX.data_path()` functions? They also fetch datasets if they are not available, right?
- Then there is `mne.datasets.eegbci.load_data()` and `mne.datasets.limo.load_data()` – why not `.data_path()`?

Maybe there are more inconsistencies or things needing clarification, but that's all I could find for now. Thoughts?

hoechenberger, 2024-08-02:
> * Then there is `mne.datasets.eegbci.load_data()` and `mne.datasets.limo.load_data()` – why not `.data_path()`?

I think these quite literally load some data, but I'm not exactly sure

cbrnr, 2024-08-02:
> I think these quite literally load some data, but I'm not exactly sure

Yes, but it just returns the path like all the other `.data_path()` methods. From the [docs](https://mne.tools/stable/generated/mne.datasets.eegbci.load_data.html#mne-datasets-eegbci-load-data): "Get paths to local copies of EEGBCI dataset files. This will fetch data for the EEGBCI dataset, which is also available at PhysioNet."

drammock, 2024-08-02:
Some datasets have the option of only downloading certain participants or runs. IIRC those are the ones with "load_data" funcs. Anything where we always download the entire dataset (e.g. because it's stored as a monolithic tar file) gets a "data_path" instead

hoechenberger, 2024-08-02:
i'd suggest that for now, let's limit the scope of this issue to `fetch_fsaverage()` please.

larsoner, 2024-08-02:
Argh must have missed it in #10348. +1 for changing, but for backward compat we should probably do what we did there which was make use of specialized Path subclasses to allow people to keep doing `+` for one release:

https://github.com/mne-tools/mne-python/pull/10348/files#diff-a176c176c64bba7aedc67f81f2ab701041ac293e7c1f6305f8a49e536cf97f1dR529

@hoechenberger are you up for doing this? Would be good to get it into 1.8

hoechenberger, 2024-08-02:
@larsoner I don't think I'll have time to tackle this within the next 2 weeks, sorry!

larsoner, 2024-08-02:
Okay I'm on it then

PR #12771 (2024-08-04): API: Return Path in fetch_fsaverage
Closes #12768

Transitioned some testing and datasets code to `pathlib.Path` while I was in there tinkering anyway.
closed 2024-08-04 by hoechenberger