mne-tools/mne-python #12772: Add Missing Fields to RawEEGLAB and EpochsEEGLAB Classes

MohammadMow, 2024-08-02:
### Describe the new feature or enhancement

The current implementation of the `RawEEGLAB` and `EpochsEEGLAB` classes in the MNE library does not load all available fields from EEGLAB .set files. Important fields such as 'icaact', 'icawinv', and several others are missing. This feature request aims to extend the functionality of these classes to include all possible fields present in EEGLAB .set files.

The following fields should be added to both `RawEEGLAB` and `EpochsEEGLAB` classes:
- `icaact`
- `icawinv`
- `icasphere`
- `icaweights`
- `icachansind`
- `chanlocs`
- `urchanlocs`
- `chaninfo`
- `ref`
- `event`
- `urevent`
- `eventdescription`
- `epoch`
- `epochdescription`
- `reject`
- `stats`
- `specdata`
- `specicaact`
- `splinefile`
- `icasplinefile`
- `dipfit`
- `history`
- `saved`
- `etc`
- `run`
- `roi`
- `datfile`

### Describe your proposed implementation

###  Using Composition and Create a DataContainerEEGLAB Class

I think `Composition` is a good solution. This class will be responsible for storing and managing all the fields from the EEGLAB .set file. The class will be initialized with a Bunch object containing EEG data and will extract and store all possible fields. Then `RawEEGLAB` and `EpochsEEGLAB` classes will have a `data_container` field. 

### Describe possible alternatives

###  Multiple Inheritance

I think it is better not to use multiple inheritance because it may cause problems such as having the same name of the methods, etc.

mscheltienne, 2024-08-08:
The ICA fields are read by this function in an ICA object: https://mne.tools/dev/generated/mne.preprocessing.read_ica_eeglab.html
Or is something else missing?

drammock, 2024-08-08:
closing as not planned; see discussion in https://github.com/mne-tools/mne-python/pull/12773#issuecomment-2275950319

closed 2024-08-08 by drammock