mne-tools/mne-python #12785: Replace `psutil.virtual_memory()` with our own implementation?

cbrnr, 2024-08-13:
Currently, `mne.sys_info()` requires the `psutil` package to list the total amount of installed memory. Unfortunately, `psutil` is not a core dependency, but only part of the `[full]` extras, which installs a large number of packages.

Since we probably don't want to promote `psutil` to a core dependency (it is not pure Python), what do you think of rolling our own solution? This could be as simple as:

```python
import platform
import subprocess

def get_total_memory():
    total_memory = "?"

    if platform.system() == "Windows":
        output = subprocess.check_output("systeminfo", shell=True).decode()
        for line in output.splitlines():
            if "Total Physical Memory" in line:
                total_memory = int(line.split(":")[1].strip())
    elif platform.system() == "Linux":
        output = subprocess.check_output(["free", "-m"]).decode()
        lines = output.splitlines()
        mem_info = lines[1].split()
        total_memory = int(mem_info[1])
    elif platform.system() == "Darwin":
        total_memory = int(
            subprocess.check_output(["sysctl", "hw.memsize"])
            .decode()
            .split(":")[1]
            .strip()
        )
        total_memory = int(total_memory / 1024**2)

    return f"Total Memory: {total_memory} MB"

print(get_total_memory())
```

I have only tested this on Linux so far (but I will run it on macOS and Windows soonish). WDYT?

larsoner, 2024-08-13:
Sure, assuming we wrap some stuff in a `try/except` so that we end up with the `?` in case there is some problem with the calls (it's not critical info and we don't want `mne sys_info` to ever fail really)

cbrnr, 2024-08-13:
OK, PR incoming!

closed 2024-09-11 by larsoner