mne-tools/mne-python #12791: Add sensor size option for plotting iEEG electrodes

AlexLepauvre, 2024-08-14:
### Describe the new feature or enhancement

When plotting iEEG electrodes on brain surfaces using [mne.viz.plot_alignment](https://mne.tools/dev/generated/mne.viz.plot_alignment.html) or [mne.viz.Brain](https://mne.tools/dev/generated/mne.viz.plot_alignment.html) add_sensors method, the electrode size is set to a default. In some cases, being able to set sensor to custom sizes would be handy for visualization. Ideally, we should be able to set each sensors to specific sizes. 

### Describe your proposed implementation

A new argument could be added to the [_plot_sensors_3d](https://github.com/mne-tools/mne-python/blob/ac863e63e6233dd3ddb60e1058731f1839e55b02/mne/viz/_3d.py#L1456) called `sensor_scale`, which takes either a single value to set all electrodes to the same size, or an array of floats specifying the size of each sensor to be plotted. This argument can be exposed in [mne.viz.plot_alignment](https://mne.tools/dev/generated/mne.viz.plot_alignment.html) and plot_sensors of [mne.viz.Brain](https://mne.tools/dev/generated/mne.viz.plot_alignment.html). We can specify it in the same way as sensor_colors

### Describe possible alternatives

I am not if there is another option available.

larsoner, 2024-08-14:
Seems reasonable to me! Especially if we can refactor some of the logic from `sensor_colors` handling to handle `sensor_sizes` in a sensible way then it hopefully won't add much additional code complexity either.

AlexLepauvre, 2024-08-14:
Perfect, I will get to it then and submit a PR once it is implemented.

closed 2024-08-23 by AlexLepauvre