mne-tools/mne-python #12818: code credit dedup

drammock, 2024-08-27:
Some possible deduplications:

- `Jean-Remi KING` and `Jean-Remi King` are separate entries
- `Valerii` and `Valerii Chirkov`
- `PAB` and `Pierre-Antoine Bannier`
- `Jordan Drew` and `Drew, J.`
- `Deleted user` still shows up (even though it's trying to be excluded in `update_credit_rst.py`)

Something I think we discussed before, but I can't recall what was decided:

- there are some folks who I'm pretty sure are in our `.mailmap` but their usernames aren't getting parsed out here (`TzionaN`, `Sena`, and `Martin` are a few examples I noticed; probably there are more). Did we decide not to use `.mailmap` for some reason? Or is it a bug that these users aren't getting the preferred name displayed?

larsoner, 2024-08-28:
Regarding names when things were ambiguous I think I tried to side with names.inc`. There are three potential sources of truth: that, GitHub profile, mailmap. I figure mailmap is most remote from the users (maintained by us) so we should probably try to make it match the other two (esp. names.inc since users typically had to make a conscious choice at time of contribution for that) but not 100% sure.

larsoner, 2024-08-28:
I should be able to look into this tomorrow

PR #12820 (2024-08-30): MAINT: Fix bugs with code credit and linking
Closes #12817
Closes #12818

Also fix the neo part of https://app.circleci.com/pipelines/github/mne-tools/mne-python/25126/workflows/44f5e760-f46a-4e48-8a46-775135dcee05/jobs/68338, need to wait for Dipy to push out new wheels for other fixes.
closed 2024-08-30 by larsoner