mne-tools/mne-python #12854: > I think the only cases where we should worry about this are in MNE itself. End users should not mess with private attributes like this. And if they do, they should expect things to break silently in unpredictable ways at any time. If we start trying to protect against this, we end up in an unsustainable rabbit hole and break a policy/contract we've always tried to follow.

larsoner, 2024-09-16:
> I directly set this [filenames] attribute [...] because I find setting up a new reader by adding a new class a bit cumbersome. I mostly do the file-specific stuff and then create a `RawArray`, so maybe adding a public setter in that class might be an option?

_Originally posted by @cbrnr in https://github.com/mne-tools/mne-python/issues/12843#issuecomment-2352441225_

cbrnr, 2024-09-16:
Perfect, that's really great indeed! I'll switch to this approach then for my readers. Thanks!

mscheltienne, 2024-09-17:
Maybe it could be turned into an example?

cbrnr, 2024-09-17:
FWIW, what's really missing in the example is a function `read_raw_xxx()`, because that's what's user-facing. This generates a bit of an overhead and redundancy (and this was the original reason why I took the shortcut), but it's perfectly OK having to implement both a class and a function (or maybe not perfectly OK, but changing the whole reader implementation would not be worth it I guess 🤷).

drammock, 2024-09-17:
> Maybe it could be turned into an example?

exactly what I was thinking; I'll re-open and re-title this issue as an issue about doc improvement.

> FWIW, what's really missing in the example is a function `read_raw_xxx()`, because that's what's user-facing.

Agreed. I think the new/edited example would be most useful if it were something like:

```py
# If no reader exists yet for the kind of file you're working with,
# here's a template function that should get you up and running:

def read_raw_npz(...):
    # load the data (this part will usually be different for each file type)
    # create fake info (details of how the channel names/types are stored in the file may vary)
    # use RawArray to instantiate
    # return RawArray

# (now show it in action)
foo = read_raw_npz(...)

# If you want to contribute your file reader to MNE-Python, it would need a few more pieces,
# like supporting `preload=False`. Here's an expanded reader for `.npz` to illustrate what a
# full-fledged reader function usually looks like:

def read_raw_npz(...):
    ...

# (now show it in action)
foo = read_raw_npz(..., preload=False)
```

cbrnr, 2024-09-17:
I'm working on this here: https://github.com/cbrnr/mnelab/pull/434

I think the `.npy` reader could be used for the example, but I don't think I will have time to add the "few more pieces". I'm not sure if it is really necessary to show everything, because then we might as well add the reader to our codebase, or no?

cbrnr, 2024-09-17:
Also, in your example I think subclassing `BaseRaw` is really essential, otherwise there's no way to set the `.filenames` attribute. This is what brought us here in the first place.

drammock, 2024-09-17:
> I think the `.npy` reader could be used for the example, but I don't think I will have time to add the "few more pieces". I'm not sure if it is really necessary to show everything, because then we might as well add the reader to our codebase, or no?

I think `.npz` (or `.hdf5` for that matter) are good candidates for "example in our docs, but not in our API" because they're user-customizable formats that aren't guaranteed to follow a proper "spec" in terms of what the different elements are named, their order, etc. So in that sense even building out a "full" reader function for them in the docs only seems fine to me.

Aside: I think `.npy` is not a good candidate for an example because it can only contain one array (e.g., only the data, not sfreq, ch_names, ch_types, etc; unless you did an object array I guess).

drammock, 2024-09-17:
> Also, in your example I think subclassing `BaseRaw` is really essential, otherwise there's no way to set the `.filenames` attribute. This is what brought us here in the first place.

ah, right you are.

cbrnr, 2024-09-17:
> I think `.npz` (or `.hdf5` for that matter) are good candidates for "example in our docs, but not in our API" because they're user-customizable formats that aren't guaranteed to follow a proper [...]

I agree!

> Aside: I think `.npy` is not a good candidate for an example because it can only contain one array (e.g., only the data, not sfreq, ch_names, ch_types, etc; unless you did an object array I guess).

Right, `.npz` would be a better candidate for an example. I've still implemented a reader for `.npy`, because it was requested and I guess some people use it because it's so simple to just dump an array into it. Of course, additional information needs to be provided elsewhere (it's in a dialog in my case).