mne-tools/mne-python #12855: CURRY Data Format Reader only works in specific cases

CurryKaiser, 2024-09-16:
### Description of the problem

Dear MNE Team!

I am from the Curry Dev team and our users often ask about your Curry reader. We believe that it isn't up to date / complete and would like to offer some help (if wanted).

Attached you can find a python script (.py but hiding as .txt) that we use (and maintain) to load Curry data into python. Maybe the resulting data structure can be used by you to transform it according to MNE conventions.

[curryreader.txt](https://github.com/user-attachments/files/17016202/curryreader.txt)
[README.txt](https://github.com/user-attachments/files/17016203/README.txt)
[requirements.txt](https://github.com/user-attachments/files/17016204/requirements.txt)

If you need any test data, we would be happy to provide! We know that this might not be of highest importance, so we would understand if it isn't something that can be done at the moment.

Best Regards,
Fabian Kaiser
Compumedics - Curry Dev Team

### Steps to reproduce

```Python
N/A
```

### Expected results

N/A

### Actual results

N/A

### Additional information

N/A

agramfort, 2024-09-19:
hi @CurryKaiser, yes can you share a super tiny file eg 5MB so we can use it for testing when integrating this code? thanks

CurryKaiser, 2024-09-20:
Yes of course! Actually, I have prepared a few files for you:
[SampleFiles.zip](https://github.com/user-attachments/files/17073979/SampleFiles.zip)
Let me know how it goes. Hope it isn't too much of a hassle and thank you for all your efforts!

agramfort, 2024-09-21:
thanks. Can you clarify what are files .cdt.dpo vs .cdt.dpa? I only see issues with the file MEG_A + MEG_B + Oth.cdt as it has a .dpo file (no .dpa)? Is it a different curry version?

CurryKaiser, 2024-09-22:
of course. .dpo and .dpa files are parameter and sensor geometry files. .dpo files are written at the time of aquisition and .dpa files are written when the .cdt files are opened for analysis. .dpa takes precedence over .dpo if it exists and this is important to keep the original data and parameters.

Maybe it could be worth considering to base the MNE Curry reader on our Python reader i.e. calling it first then just converting the output into MNE file format. This way whenever we add something to our file format, your reader would be up to date as well, as we maintain the Python reader. Though it might be a little early to tell if that is less work.

A collegue of mine has briefly skimmed the code of your Curry reader in the past and noticed some bugs and maybe some things missing but he never got around to reporting it and quite some time has passed. It would be a relief though if it is just the .dpo file that couldn't be read (as it is very similar to a .dpa file).

agramfort, 2024-09-22:
yes we could base our reader on your code. For packaging reasons, we would
need to have your code in a released package in pypi and conda (although we
can help with that). Would it be possible for you to do this? Also when you
find bugs in our code please open an issue in our tracker so action can be
taken asap. thanks

> Message ID: ***@***.***>
>

CurryKaiser, 2024-09-24:
It might be possible. But I wanna hold off on commenting any further until our Python Reader expert is back which will be in 2 weeks.
Yeah sorry about the unreported bug, it has been a long while ago and I am sure my collegue was swamped with work at that time.

CurryKaiser, 2024-10-17:
Apologies for the delay.

These were the issues so far that our expert did spot:

1) Missing "cdt.dpo"
[image]

2) Landmark tokens:
[image]

3) Issue with dev_head_transform:
[image]

I will get back to you with pypi and conda packages. I think that  should prevent any headaches in the future.

CurryKaiser, 2024-10-17:
At this point, it is unsure whether we can release the packages in the way that you suggest. I will keep you up to date when a decision is made.

CurryKaiser, 2024-11-13:
Unfortunately, we cannot. But our latest reader is always on github: https://github.com/neuroscan/curry-python-reader
Thank you for your time!i

drammock, 2025-03-31:
@CurryKaiser you probably saw that one of our contributors (@dominikwelke) has started integrating your https://github.com/neuroscan/curry-python-reader into MNE-Python over in https://github.com/mne-tools/mne-python/pull/13176. After some further discussion over there, we seem to be coalescing on a plan to:

1. fork your reader repository
2. add what is necessary to make it an installable Python package
3. package and distribute it via PyPI and conda-forge
4. depend on that package in MNE-Python, use it to do the low-level Curry reading operations, and write some wrapper code around it to create MNE-Python objects.

[in your comment above](https://github.com/mne-tools/mne-python/issues/12855#issuecomment-2473394145), you indicated you aren't willing to package it yourselves (with or without our help). Since the code is BSD-licensed I guess we would be within our rights to do steps 1-4 regardless, but I wanted to keep you in the loop before we take those steps, in case anything has changed on your end. Would you still prefer that the repo `neuroscan/curry-python-reader` be left up to your people, and not be the source of an installable curry reader package?  If so, we'll move forward with packaging/distributing/etc as described above.