mne-tools/mne-python #12864: API: Remove raw.impedances

larsoner, 2024-09-19:
See discussion in https://github.com/mne-tools/mne-python/issues/12861#issuecomment-2361102202

larsoner, 2024-09-26:
Argh @sappelhoff I went to look at `pybv` and the code there seems to have to do almost exclusively with writing, not reading BV data.

@sappelhoff @WDYT about just keeping the code in MNE-Python with a `mne.io.brainvision.read_impedances(...)`?

drammock, 2024-09-26:
obligatory cross-ref to https://github.com/mne-tools/mne-python/pull/12450#issuecomment-1967238036, in particular

> If the desired end-state here is "cleaner separation of concerns" / "pybv only handles brainvision stuff, not MNE stuff" then I think there's a strong case that pybv should absorb MNE-Python's brainvision *reading* code

larsoner, 2024-09-26:
Oh okay with me then!

drammock, 2024-09-26:
> Oh okay with me then!

unfortunately my suggestion in https://github.com/mne-tools/mne-python/pull/12450#issuecomment-1967238036 was not met with enthusiasm. But maybe now the case is stronger / more pursuasive?  WDYT @cbrnr @sappelhoff?

cbrnr, 2024-09-27:
I'd be OK with adding reader functionality to pybv, it does make a lot of sense. However, this would be the first format (I think) where we choose to do so, so why is this a problem suddenly? If the current reader function does read impedances (if present), and the only reason we're now removing it is because it doesn't survive a round-trip to FIFF, then IMO the problem is that we're still relying on FIFF. I'm pretty sure we've had this discussion before, but no one outside the small MEG community has ever used FIFF. It's a very niche file format that doesn't even have many readers outside of MNE. So why don't we think about moving to a new internal format that is independent of the FIFF idiosyncrasies, such as HDF? I think it could be as easy as dumping our current `Raw` into HDF, and then we'd be free to extend attributes even if FIFF does not support it.

cbrnr, 2024-09-27:
PS: Even for EDF we're rolling our own reader, even though edfio includes a reader and a writer. I'm not saying it is a bad thing that we implement our own readers (after all, we get to decide how strict they follow a given standard, which is not always desirable), so for me readers are not really up for discussion. It's the FIFF format that should be swapped with something better.

mscheltienne, 2024-09-27:
> IMO the problem is that we're still relying on FIFF. I'm pretty sure we've had this discussion before, but no one outside the small MEG community has ever used FIFF. It's a very niche file format [...]

Agree, but that would be a major and complicated refactor to do.

sappelhoff, 2024-09-27:
> Argh @sappelhoff I went to look at pybv and the code there seems to have to do almost exclusively with writing, not reading BV data.

Yes, that is true. I thought we'd want to very selectively outsource impedance reading to `pybv`, which would not be so hard to do. But it is true that is would be a bit odd that no reading is supported EXCEPT for impedances

> @sappelhoff WDYT about just keeping the code in MNE-Python with a mne.io.brainvision.read_impedances(...)?

would be fine with me, too -- as long as we retain SOME way to get the impedances.

> unfortunately my suggestion in https://github.com/mne-tools/mne-python/pull/12450#issuecomment-1967238036 was not met with enthusiasm. But maybe now the case is stronger / more pursuasive? WDYT

I would be fine with having pybv take over READING, as long as it supplies the read data in a "non-MNE" format (basic Python and numpy data structures), to stay "lightweight/minimal", and then have code in mne that turns these basic data structures into mne-ready structures.

However, it would require some work and effort that I cannot commit to at this point.

> IMO the problem is that we're still relying on FIFF. I'm pretty sure we've had this discussion before, but no one outside the small MEG community has ever used FIFF. It's a very niche file format [...]

I agree with Clemens and Mathieu, the FIFF format often places an unnecessary limit on our plans of supporting more (or more detailed) (meta-) data. I am also aware that a potential change of the "saving backend" of mne would probably be the biggest change (labor intensive, error prone, ... but also valuable, and future-oriented) MNE-Python has seen in its current lifetime.

cbrnr, 2024-10-07:
> I agree with Clemens and Mathieu, the FIFF format often places an unnecessary limit on our plans of supporting more (or more detailed) (meta-) data. I am also aware that a potential change of the [...]

It will be a big change, yes, but I'm not even sure if it's going to be that problematic if we start by just dumping what we have into an HDF5 container. There are of course some details to this process, but if people think this would be worth discussing, I suggest that we open a new issue.

Regarding the removal of impedances, I think there's not really a strong need to do that, or is there? Do we document which attributes survive a round trip through FIFF? If `raw.impendances` cannot/should not be used, then what about at least providing it in `raw._extras` (at least as long as FIFF remains the standard format)?

cbrnr, 2024-10-07:
> First I'll say that this FIF limitation has come up before previously, and the consensus/decision that I recall was that we should have `mne.export` for other formats. So I'd rather not work through all those discussions if possible. I'd like to stick with `raw.save` meaning "to FIF" and `mne.export` be for everything else.

Fair enough, but this is hopefully not set in stone, so I'd rather discuss it again from time to time. This means it is bothering people enough apparently that it might be worth reconsidering.

Your remaining arguments miss the point I think. We can also have a very well defined interface even if we switch to HDF5. The actual storage format is just an implementation detail, and I'd rather have a standardized format like HDF5 than a very specific format that basically only MNE can read. But yes, if you're saying we can adapt the standard, I'd vote for adding an impedance field to FIFF then.

Regarding implementation effort, of course someone has to invest the time, but depending on how flexible HDF5 is, I imagine that it will be possible to dump our current format into HDF5, which shouldn't be that big of a deal.

cbrnr, 2024-10-07:
Also, since FIFF is well specified, where can I find the specifications if I wanted to implement a FIFF reader or writer from scratch?

larsoner, 2024-10-09:
> Fair enough, but this is hopefully not set in stone, so I'd rather discuss it again from time to time. This means it is bothering people enough apparently that it might be worth reconsidering.

Of course nothing is set in stone, that would not be good! But on the other hand, for bigger decisions like this, typically multiple people spend a lot of time (days? weeks? months?) discussing, thinking, and coming to a consensus and implementing the agreed upon solution. I think it's critical to take that into account, and only if there is quite strong support and evidence that the current solution is problematic reverse the decision. Gathering this evidence would require looking back into all the old discussions, PRs, and issues as much as possible and understanding all the points there (which isn't clear has been done yet?), and why they are convincingly, sufficiently wrong or problematic now that we should change. If every time we make a big decision, a year or two later someone goes "I (still) don't like that decision" and we are required to rehash all the old discussions and pros and cons, I don't think it's a good use of project time or effort, or really fair to the people who put in that time and effort to have the original discussions. Yes if it's problematic let's fix it but if it's not clearly wrong let's move on to other stuff!

For FIFF specification I wouldn't claim it's super well documented like some other formats like EDF. I think the info lives to some extent in fiff-constants but also (maybe moreso) in code in MNE-Python and MNE-MATLAB.

sappelhoff, 2024-10-09:
> For FIFF specification I wouldn't claim it's super well documented like some other formats like EDF. I think the info lives to some extent in fiff-constants but also (maybe moreso) in code in MNE-Python and MNE-MATLAB.

Out of curitosity, and because it is relevant to this discussion: Is FIFF not a vendor specific format by Elekta / Neuromag / MEGIN (in historical order of company names)? If we make changes to the "FIFF spec", are we talking about some official file format specification that is then also respected by every other user? Or are we talking about an mne specific fork of the FIFF spec?

I cannot find traces of it anymore, but I remember there being a discussion on how MNE-Python-produced FIF files may not always be read by FIF readers that CAN read "out-of-the-machine-FIFs".

larsoner, 2024-10-09:
> @larsoner I've never seen this discussed here, except maybe in #5302, but there hasn't been any definitive answer.

You should look more broadly into discussions of `raw.save` vs `mne.export`. I would start from the `git blame` of `mne.export` and you'll see some original PRs which should then link to discussions hopefully. I know some of the discussions were not on GitHub and instead in person (possibly at sprints?) but at least some of them should be discoverable still.

> Why do you think switching to HDF5 to dump our current raw structure would be so much work? We don't change anything, just the file format.

I don't think that part would be a ton of work, but I do think based on previous discussions that if we do it it should live in `mne.exports`. And I haven't seen much (any?) demand for being able to write data to HDF5 this way. I assume people mostly use `pymatreader` for compat with MATLAB mostly. Critically I am not sure what other software (besides MNE if we wrote a `mne.io.read_raw_h5`) would *read* this output, so from a compat standpoint I don't think it gains much.

> Lastly, we need to find a solution for the raw.impedances issue. I think we should not remove this functionality just because FIFF doesn't support it.

I think we should either add it to the FIF spec, which would be okay, or instead (easier, seems okay to me, agreed upon previously by all others I thought?) move it to pybv and antio for now. And if users clamor for this info to be stored more permanently and is more widely available in other formats (which I don't know?) then we move to support it in FIF, and in mne.io.Raw concurrently.

cbrnr, 2024-10-09:
You're really avoiding the issue of FIFF being a proprietary format. Is this not an issue for everyone here?

drammock, 2024-10-09:
Since you linked to #5302 you probably saw this already, but in case not: [here is the FIFF file format spec](https://www.dropbox.com/s/q58whpso2jt9tx0/Fiff.pdf?dl=1)

cbrnr, 2024-10-09:
Of course I saw the link to some random Dropbox file. This is not what I'd say an open file format documentation should be like (never mind that it's from 2011 and that it contains copyrighted material, so I can't be sure that the document is even meant to be public).

sappelhoff, 2024-10-09:
> We now co-maintain the spec officially at https://github.com/mne-tools/fiff-constants.

Thanks! I didn't know this.

>  Please open a separate issue for improving documentation of the FIFF file format, and let's keep this thread focused on the impedances issue.

+1 for three separate discussions / action items:

1. improving FIFF documentation, and potentially a page somewhere that transparently summarizes the motivation of mne to rely on FIFF (historical or otherwise)
2. discussion on where to put impedances (outsource to pybv/etc., new FIF constant, ...)
3. discussion on potential future changes to the main mne IO backend

mscheltienne, 2024-10-09:
> Again, FIFF is a proprietary format, and maintaining just some constants is not a file format documentation. I don't think that an open-source package should be based on a proprietary format.

Personaly, I don't see the issue even if the format was proprietary. If the format is well made and suits our need perfectly, why not use it? But again, as @drammock mentions, it's not even proprietary, it's an open standard, not that well documented, but open.

> Why do you think switching to HDF5 to dump our current raw structure would be so much work? We don't change anything, just the file format.

I'm -1 for this. As @larsoner says, if we wanted to support a different saving/export format, we could do it in `mne.Export`, but HDF5 is *too* flexible for this. There is no gain in saving to a format that everyone else will be able to read but no one else will be able to parse without writing a good amount of code.

IMO, dumping our current raw structure in a different format serves no purpose. I am more bothered by the structure of our objects `mne.Info`, `mne.io.Raw`, ... deriving heavily from the FIFF format. As more and more formats were read into MNE-Python, we always found a way to fit whatever information we could parse from those files into our containers.. but sometimes you need to get creative or eventually amend the FIFF specs when you can't find a sensible way to make it fit. It adds a level of complexity and of 'mess' in the code that could be avoided with a better thinking of what our containers should look like and what they should accomodate. But that's a project for another day 😉 

> Lastly, we need to find a solution for the raw.impedances issue. I think we should not remove this functionality just because FIFF doesn't support it.

+1 to remove it from MNE for now, while preserving reading in the I/O packages with an example in the MNE documentation. The current implementation was made for the brainvision format without even thinking about possible other formats, which might offer multiple impedance recordings within a single file. So now if we want to add this second case, we have a disparity between what `raw.impedances` contains for BV and for these other formats.

Alternatively, we could think about how to support multiple impedance measurements / recording within the FIFF specification and our structure, but I am not convinced that this is that useful and requested by our users.

-------------------

Overall:

> 1. improving FIFF documentation, and potentially a page somewhere that transparently summarizes the motivation of mne to rely on FIFF (historical or otherwise)

+1 especially on the 'improving FIFF doc' part

> 2. discussion on where to put impedances (outsource to pybv/etc., new FIF constant, ...)

+1 to move the impedance parsing to I/O packages with an example on our documentation

> 3. discussion on potential future changes to the main mne IO backend

-- no opinion.