mne-tools/mne-python #12873: Raw plot yields either empty or buggy plot

Anj-RU, 2024-09-26:
### Description of the problem

I'm working with intracranial EEG data. The plotting is erratic. Sine .fif files yield a blank screen if plotting from jupyter notebook (.ipynb). If plotting from a .py file, the data looks corrupted no matter what uV I choose. 

Interestingly, some .fif files seem to load fine, so I'm not sure if it's something inherent to the data itself that is problematic. However, the data in the actual object looks fine if I plot it using matplotlib.

.ipynb (tested with vscode):
[image]

.py (tested with vscode and pycharm):
[image]

### Steps to reproduce

```Python
#The code for the .ipynb:
raw_file = mne.io.read_raw_fif(fname, verbose=True,preload=False)
raw_file.load_data()
raw_file.notch_filter(np.arange(60, 241, 60)).filter(l_freq=1., h_freq=None)
raw_file.plot(scalings = dict(mag=1e-12, grad=4e-11, eeg=350e-6, eog=150e-6, ecg=5e-4,
     emg=1e-3, ref_meg=1e-12, misc=1e-3, stim=1,
     resp=1, chpi=1e-4, whitened=1e2))

#The code for the .py:
raw_file = mne.io.read_raw_fif(fname, verbose=True,preload=False).crop(tmin=0*60, tmax=10*60) #.pick(np.arange(20,50)) #.crop(tmin=420*60, tmax=500*60).resample(512)
raw_file.load_data()
raw_file.plot(scalings = dict(mag=1e-12, grad=4e-11, eeg=35e-6, eog=150e-6, ecg=5e-4,
     emg=1e-3, ref_meg=1e-12, misc=1e-3, stim=1,
     resp=1, chpi=1e-4, whitened=1e2))
plt.show()
```

### Expected results

I expect a good looking plot

### Actual results

The plot looks bad

### Additional information

Platform             macOS-14.6.1-arm64-arm-64bit
Python               3.11.7 | packaged by conda-forge | (main, Dec 23 2023, 14:38:07) [Clang 16.0.6 ]
Executable           /Users/michaa08/Library/Mobile Documents/com~apple~CloudDocs/Python_Scripts/Environments/eegPreprocess/bin/python
CPU                  arm (12 cores)
Memory               64.0 GB

Core
├☑ mne               1.8.0 (latest release)
├☑ numpy             1.26.3 (OpenBLAS 0.3.26 with 12 threads)
├☑ scipy             1.12.0
└☑ matplotlib        3.8.2 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.4.0
├☑ numba             0.58.1
├☑ nibabel           5.2.0
├☑ nilearn           0.10.2
├☑ dipy              1.7.0
├☑ openmeeg          2.5.7
├☑ pandas            2.2.0
├☑ h5io              0.2.1
├☑ h5py              3.10.0
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.43.2 (OpenGL 4.1 Metal - 88.1 via Apple M2 Max)
├☑ pyvistaqt         0.11.0
├☑ vtk               9.2.6
├☑ qtpy              2.4.1 (PyQt5=5.15.8)
├☑ pyqtgraph         0.13.3
├☑ mne-qt-browser    0.6.1
├☑ ipywidgets        8.1.1
├☑ trame_client      2.15.0
├☑ trame_server      2.15.0
├☑ trame_vtk         2.7.0
├☑ trame_vuetify     2.4.2
└☐ unavailable       ipympl

Ecosystem (optional)
├☑ eeglabio          0.0.2-4
├☑ mffpy             0.8.0
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, edfio, pybv

agramfort, 2024-09-26:
your scaling is definitely off. You can press "-" key to reduce scaling with qt browser and matplotlib if it's a qt window. You can also look at the "auto" parameter value for scalings.

Anj-RU, 2024-09-26:
I've tried both of those, looking at scaling between 0uV to 7000 uV. In
either of the above UIs, it doesn't change the appearance at all.

agramfort, 2024-09-26:
Can you tell me the standard deviation of your data in the raw object?

Anj-RU, 2024-09-26:
Ah you've solved it, thank you! The general stdev range across the 96 channels is 40-160 with a few outliers...

So I changed eeg=350e-6 to eeg=40, and this did the trick. The voltage marker in the UI now says 125000000.0 uV, but the EEG looks good.

The majority of the files in this dataset are visualized well at 700uV (i.e. eeg=350e-6), so I'm wondering if there was some conversion that was/wasn't done for these other files. Thank you very much!!!!

closed 2024-09-26 by Anj-RU