mne-tools/mne-python #12876: How to cancel the labeling of trigger channel

baili-b7ack, 2024-09-27:
I was using Curry 8 along with E-Prime 3, and was trying to add customized labels to eeg scroll. But it seems that with trigger channel, it has its own way of labelling events, and those labels overlapped with the label I set up and the scroll looks messy. (like the pic given)
[image]

Is there a way to stop the automatic labelling of trigger channel?

mscheltienne, 2024-09-27:
Hello, I do not understand what you mean by automatic labeling, or how you got this plot.

mscheltienne, 2024-09-27:
This plot is not related to MNE-Python? Neither is any of the parts you are mentionning so far?

baili-b7ack, 2024-09-27:
Oops, I'm sorry, I don't know where to submit this and while googleing for answer I found issues seemingly related to Curry software in this project, so I submitted this issue. Sorry for interrupting. ><

baili-b7ack, 2024-09-27:
> This plot is not related to MNE-Python? Neither is any of the parts you are mentionning so far?

Oops, I'm sorry, I don't know where to submit this and while googleing for answer I found issues seemingly related to Curry software in this project, so I submitted this issue. Sorry for interrupting. ><

drammock, 2024-09-27:
If your plot is generated by the curry reader package, try submitting the issue here: https://github.com/neuroscan/curry-python-reader

closed 2024-09-27 by drammock

baili-b7ack, 2024-09-28:
> If your plot is generated by the curry reader package, try submitting the issue here: https://github.com/neuroscan/curry-python-reader

Thanks, I'll try