mne-tools/mne-python #12881: Support for Neuro-1 OPM System

neurofractal, 2024-10-01:
### Describe the new feature or enhancement

Read in .lvm data from the Neuro1 OPM system from QuSpin Inc.

### Describe your proposed implementation

Write mne.io.read_raw_neuro1 in the same vain as mne.io.read_raw_fil. The documentation for .lvm file format is described [HERE](https://github.com/user-attachments/files/17155805/QuSpin_Data_File_Format_Description.pdf). Also write a tutorial with some interesting example data (e.g. auditory tones?)

### Describe possible alternatives

N/A - new feature

### Additional context

See #8671 and https://github.com/fieldtrip/website/issues/807

dbobela406, 2025-01-09:
Hi neurofractal, 

Reaching out to get an update on this feature--what is the latest progress? Cheers!

dbobela406, 2025-01-14:
> Hey - so the Cerca system has an inbuilt lvm/cMEG to fif converter so not
> a priority our end. Good luck!

Thanks for the update! I'll plug away at building a quspin loader. Cheers,
Dave