mne-tools/mne-python #12894: Detect badly scaled plots in raw.plot, warn and propose auto-scale

mscheltienne, 2024-10-11:
### Describe the new feature or enhancement

Again on the forum: https://mne.discourse.group/t/raw-plot-shows-an-empty-figure/10371
This kind of scaling problem is common, I remember seeing it several times on the forum. How about we:

- detect data which is likely badly scaled when calling `raw.plot()`
- issue a warning as a **pop-up window**
- propose in this **pop-up window** to auto-scale the data  

### Describe your proposed implementation

-

### Describe possible alternatives

-

larsoner, 2024-10-15:
Agreed this is probably most easily doable in the Qt backend. +1 for moving it to `mne-qt-browser` and someone can implement it when they have time!

It would also be good to be able to disable this pop-up window using an MNE-Python config var, as I could see how for some workflows people might *know* their data scales yield huge traces and it would be bad if they always got an unnecessary popup.

mscheltienne, 2024-10-16:
> I hesitate a lot about adding pop-ups like this. Is a (big, bold, all-caps?) warning in the terminal not sufficient?

In my experience, people easily miss it when they have a GUI open, usually in front of the terminal.
Another option would be a small part of the GUI dedicated to 'logs' which would forward info levels logs in white, warning in red, ...