mne-tools/mne-python #12897: Error when reading an EDF file (resampling)

cbrnr, 2024-10-13:
### Description of the problem

From the [forum](https://mne.discourse.group/t/assertionerror-at-max-n-smp-read-smp-exp/10382):

> I'm trying to read a channel from an edf file using the code below:
> 
> ```python
> import mne
> edf = mne.io.read_raw_edf("my/edf/file.edf")
> out = edf.get_data(picks=["my_channel"], return_times=False)
> ```
> But the `edf.get_data` line raises an AssertionError as shown in the following traceback:
> ```
> my_file:35: in edf_data
>     out = edf.get_data(picks=["my_channel"], return_times=False)
> <decorator-gen-190>:12: in get_data
>     ???
> myvenv/lib/python3.12/site-packages/mne/io/base.py:956: in get_data
>     getitem = self._getitem(
> myvenv/lib/python3.12/site-packages/mne/io/base.py:853: in _getitem
>     data = self._read_segment(start=start, stop=stop, sel=sel)
> <decorator-gen-187>:12: in _read_segment
>     ???
> myvenv/lib/python3.12/site-packages/mne/io/base.py:472: in _read_segment
>     _ReadSegmentFileProtector(self)._read_segment_file(
> myvenv/lib/python3.12/site-packages/mne/io/base.py:2598: in _read_segment_file
>     return self.__raw.__class__._read_segment_file(
> myvenv/lib/python3.12/site-packages/mne/io/edf/edf.py:220: in _read_segment_file
>     return _read_segment_file(
> _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
>                 smp_exp = data.shape[-1]
> >               assert max(n_smp_read) == smp_exp
> E               AssertionError
> 
> myvenv/lib/python3.12/site-packages/mne/io/edf/edf.py:442: AssertionError
> 
> ```
> I cannot share the edf file but I tested with mne==1.4 and with that version works.

This might have been introduced by #11549.

### Steps to reproduce

```Python
The file cannot be shared.
```

### Expected results

There should be no error when loading this file.

### Actual results

There is an error.

### Additional information

N/A

skjerns, 2024-10-14:
The bug seems to be specific to that certain EDF-file. Without a file to reproduce I'm at loss :-/

cbrnr, 2024-10-14:
Unfortunately, I do not have the file either. The original reporter in the forum says he can't share the file, but it seems to happen with one specific channel only. However, I already mentioned that this might make it impossible for us to fix the problem.

JosephPenaQuino, 2024-10-14:
Hi, @skjerns . I made an anonymized version with two signals: "working_ch" and "failing_ch".
[output.zip](https://github.com/user-attachments/files/17368028/output.zip)

skjerns, 2024-10-15:
Can reproduce. Will look into it.

Workaround is to use `preload=True`

h-mayorquin, 2024-10-16:
commenting to follow up on this.

cbrnr, 2024-10-16:
@h-mayorquin you don't have to comment if you want to follow a PR/issue, there's a "Subscribe" button for this purpose.

PR #13069 (2025-01-17): [FIX] Reading an EDF with preload=False and mixed frequency
Fixes #12897.

When loading an EDF with mixed sample frequencies and using `preload=False` and then requesting a channel that has lower sample frequency than the maximum resulted in an `AssertionError`.

#### What does this implement/fix?

 This Assertion was in place to make sure that enough samples were read, but only made sense when one of the channels of maximum sample frequency was loaded.

It can simply be removed as I don't think that the case can actually occur. Additionally, there is another check for zeros a few lines later  [... cut]
closed 2025-01-17 by larsoner