mne-tools/mne-python #12899: MNE-Python application icon broken in Ubuntu 24.04

larsoner, 2024-10-16:
... I think due to the use of Gnome3 newer Ubuntu seems to by default now *require* a `.desktop` file to use something other than this in the dock:

[image]

I tried a bunch of Qt stuff and none of it worked. At some point I stumbled upon a post that mentioned `.desktop` with `StartupWMClass` could fix it and indeed this seems to work:

<details>
<summary>~/.local/share/applications/mne-python.desktop</summary>

```
[Desktop Entry]
Type=Application
Version=1.5
Name=MNE-Python
StartupWMClass=MNE-Python
Icon=/home/larsoner/python/mne-python/mne/icons/mne_default_icon.png
SingleMainWindow=true
NoDisplay=true
```

</details>

[image]

@hoechenberger our installers currently add files like `mne-python-180_prompt-mne.desktop` but I'm not sure if it's possible to have them create a separate linux-only file, do you?

Either way, we could consider adding a note about this to our install docs somewhere as well.

larsoner, 2024-10-17:
... looking at https://conda.github.io/menuinst/reference/ I don't see an option for `StartupWMClass` which is what we'd need to set. I'll open in issue over there since maybe we could add a Linux specific shortcut for it if we could set this.

larsoner, 2024-10-17:
Okay, opened https://github.com/conda-forge/mne-feedstock/pull/139, https://github.com/conda/menuinst/pull/265, and https://github.com/mne-tools/mne-python/pull/12900

hoechenberger, 2024-10-18:
Hello, just to clarify, no veto from my side! I only think that maybe we can come up with an additional / a better solution that requires less user interaction 👍

cbrnr, 2024-10-22:
Hi @larsoner! I have the same problem with MNELAB, but adding the `.desktop` file doesn't work for me.

I've created `~/.local/share/applications/mnelab.desktop` with the following content:

```
[Desktop Entry]
Type=Application
Version=0.9.2
Name=MNELAB
StartupWMClass=MNELAB
Icon=/home/clemens/mnelab.png
SingleMainWindow=true
NoDisplay=true
```

Do I have to do anything else? Or is it a problem that the "executable" is really running Python with a specific module (or `uvx mnelab`)?

larsoner, 2024-10-22:
You need to know the `StartupWMClass` so start Looking Glass (e.g., alt-F2 then `lg`) and inspect the window. For example for `import matplotlib.pyplot as plt; plt.figure()` (which now also has the same problem as us) in the `Windows` tab I can see:

[image]

So if I then create:

<details>
<summary>~/.local/share/applications/matplotlib.desktop</summary>

```ini
[Desktop Entry]
Type=Application
Version=1.5
Name=matplotlib
StartupWMClass=matplotlib
Icon=/home/larsoner/python/matplotlib/lib/matplotlib/mpl-data/images/matplotlib_large.png
SingleMainWindow=true
NoDisplay=true
```

</details>

It updates to use the `matplotlib` a few seconds later:

[image]

cbrnr, 2024-10-23:
Thanks @larsoner, this works! However, the `StartupWMClass` is `python3`, so any custom icon I assign would then be used for all Python windows, which is not ideal.

I assume in the case of MNE, the installer sets the class to `MNE-Python`, so it's not a problem to assign an icon. However, if that's the case, I don't understand why the installer can't add this file in the first place (I thought that you were also discussing non-installer MNE).

drammock, 2024-10-23:
> I don't understand why the installer can't add this file in the first place (I thought that you were also discussing non-installer MNE).

It will. See https://github.com/mne-tools/mne-python/pull/12900#issuecomment-2420341049. The problem is that users who don't use the installer still need it

cbrnr, 2024-10-23:
> It will. See [#12900 (comment)](https://github.com/mne-tools/mne-python/pull/12900#issuecomment-2420341049). The problem is that users who don't use the installer still need it

Right, but users without the installer will not have the `MNE-Python` class. They will have `python3`, so this workaround will not really work.

drammock, 2024-10-23:
> users without the installer will not have the `MNE-Python` class. They will have `python3`, so this workaround will not really work.

IIUC the wmclass is set at runtime by the mne qt browser package, not by the installer

cbrnr, 2024-10-23:
> IIUC the wmclass is set at runtime by the mne qt browser package, not by the installer

I don't think so. It has `python3` for me.

cbrnr, 2024-10-23:
Yeah, this sounds really brittle. I wish there was a way to set this in the app.

But I still don't understand how people who don't use the installer will benefit from the desktop icon. How can I start the Qt app by MNE-Python? What even is "MNE-Python" in this context? Sorry I keep asking questions, but maybe you can give me a simple reproducible way to open some MNE GUI app that has the `MNE-Python` window class (the CLI example `mne browse_raw ...` has `Matplotlib`).

larsoner, 2024-10-23:
For me `mne browse_raw` does it. Are you using the mne-qt-browser backend or the matplotlib backend?

This should also work in plain Python:
```
>>> import mne
>>> mne.viz.set_browser_backend("qt")
>>> raw = mne.io.read_raw_fif(mne.datasets.sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif")
>>> raw.plot()
```
This should cause mne-qt-browser to be the first thing to spawn the Qt app, which should in turn set the wmclass to MNE-Python.

larsoner, 2024-10-23:
> How can I start the Qt app by MNE-Python? What even is "MNE-Python" in this context?

For this part it's really this under the hood:
```
import mne
mne.viz.backends._utils._init_mne_qtapp()
```
That would be the QApplication created by MNE-Python (the `mne` module).

cbrnr, 2024-10-23:
> This should cause mne-qt-browser to be the first thing to spawn the Qt app, which should in turn set the wmclass to MNE-Python.

No, it's `python3` for me. I ran the commands you listed in a plain old terminal.

larsoner, 2024-10-23:
Weird, didn't know what the difference would be! But at least now you know how to change the icon if you want to.

cbrnr, 2024-10-23:
> Weird, didn't know what the difference would be! But at least now you know how to change the icon if you want to.

Well, that's kind of my point: if I change it, then *all* Python GUI apps have that same icon!

larsoner, 2024-12-09:
I think we're good enough for now here but PRs welcome to improve the situation, or we can reopen if it's still unacceptable