mne-tools/mne-python #12922: Update supporting institutions

cbrnr, 2024-10-29:
Our website contains logos of supporting institutions. We might want to update these once in a while, since contributors will change jobs from time to time. For example, I think AE Studio (which was added in #11695) should be removed, because Rob Luke does not work there anymore (according to his [LinkedIn](https://au.linkedin.com/in/rob--luke)). There are probably other logos that should be changed/removed/updated.

This also raises the question of what criteria are used to determine whether an institution's logo should be included in the list of supporting institutions - maybe something for the soon to be established Steering Council to work out.

sappelhoff, 2024-10-29:
I agree, see also:

- https://github.com/mne-tools/mne-python/pull/12896#issuecomment-2416168849

cbrnr, 2026-09-23:
Do we have a working policy in place to deal with these kinds of updates on a regular basis? If so, this issue could be closed. @larsoner @drammock

drammock, 2026-09-23:
> Do we have a working policy in place to deal with these kinds of updates on a regular basis? If so, this issue could be closed.

I don't think we do. Perhaps we should set up a cron job that opens a new issue once every 3 months (or 6 months?) as a reminder to check the list and update as needed.

I've wondered whether we should also add more detail [like sklearn does](https://scikit-learn.org/stable/institutional_support.html) by saying, e.g., "Uni Graz sponsors maintainer Clemens Brunner by providing 10% of FTE for work on open source projects" or something similar.

cbrnr, 2026-09-23:
I think checking the list twice a year should be sufficient. A periodic issue would definitely be useful! I can try coming up with something if you want.

I wouldn't bother adding more details as these things are probably pretty difficult to figure out. For example, I don't have any official mandate to work on OSS in my contract, let alone a percentage of my time. I do have fixed shares for research, teaching, and administration, but that's about it. I think I'm probably not the only one without such concrete numbers.

drammock, 2026-09-23:
It would be great if you figured out a clean way to do an every-6-months reminder issue.
I'm fine with not adding additional details (we can always add later if we change our minds).

> I don't have any official mandate to work on OSS in my contract, let alone a percentage of my time. I do have fixed shares for research, teaching, and administration, but that's about it.

in that case, does it make sense to remove Uni Graz from the supporting institutions list (or at least move to "past supporting institutions" list)?

cbrnr, 2026-09-23:
> in that case, does it make sense to remove Uni Graz from the supporting institutions list (or at least move to "past supporting institutions" list)?

I'm still contributing during my paid time, I just don't have a fixed percentage that I can dedicate to OSS contributions.

closed 2026-09-24 by larsoner