mne-tools/mne-python #12927: Error in "Spatiotemporal permutation F-test on full sensor data" tutorial

joebathelt, 2024-10-30:
### Proposed documentation enhancement

First, thank you for developing MNE-Python and for all the work that went into creating such comprehensive and useful tutorials.

I recently completed the [spatiotemporal clustering tutorial for full sensor data](https://mne.tools/stable/auto_tutorials/stats-sensor-space/75_cluster_ftest_spatiotemporal.html#compute-permutation-statistic) using MNE version 1.8. However, when I reached the clustering step, I encountered an error at Line 312 that prevented the clustering from completing successfully:

```python
Exception ignored in: 'scipy.sparse.csgraph._traversal._connected_components_undirected'
Traceback (most recent call last):
  File "/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py", line 304, in _get_components
    _, components = connected_components(adjacency)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Buffer dtype mismatch, expected 'ITYPE_t' but got 'long'
```

It seems the issue is related to the format or type of the adjacency matrix, but I’m unsure how to resolve it. Could you please advise on what might be causing this error and any steps I could take to fix it?

larsoner, 2024-10-30:
Can you paste the full traceback and also the output of `mne sys_info`? I suspect that this is a SciPy bug having to do with [`connected_components`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.sparse.csgraph.connected_components.html) but to replicate and investigate I need a bit more info!

joebathelt, 2024-10-30:
Hi @larsoner,

Thank you for the swift response. 

Here is the full traceback:
```python
stat_fun(H1): min=6.752417666025855e-07 max=38.40836555950498
Running initial clustering …
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
ValueError: Buffer dtype mismatch, expected 'ITYPE_t' but got 'long'
Exception ignored in: 'scipy.sparse.csgraph._traversal._connected_components_undirected'
Traceback (most recent call last):
  File "/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py", line 304, in _get_components
    _, components = connected_components(adjacency)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: Buffer dtype mismatch, expected 'ITYPE_t' but got 'long'
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
File /Users/joebathelt/Downloads/Spatiotemporal F-test full sensor (1).py:15
     [12](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:12) tfr_threshold = 15.0
     [14](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:14) # run cluster based permutation analysis
---> [15](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:15) cluster_stats = spatio_temporal_cluster_test(
     [16](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:16)     X,
     [17](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:17)     n_permutations=1000,
     [18](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:18)     threshold=tfr_threshold,
     [19](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:19)     tail=1,
     [20](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:20)     n_jobs=None,
     [21](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:21)     buffer_size=None,
     [22](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:22)     adjacency=tfr_adjacency,
     [23](https://file+.vscode-resource.vscode-cdn.net/Users/joebathelt/Downloads/Spatiotemporal%20F-test%20full%20sensor%20(1).py:23) )

File <decorator-gen-306>:12, in spatio_temporal_cluster_test(X, threshold, n_permutations, tail, stat_fun, adjacency, n_jobs, seed, max_step, spatial_exclude, step_down_p, t_power, out_type, check_disjoint, buffer_size, verbose)

File /opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1548, in spatio_temporal_cluster_test(X, threshold, n_permutations, tail, stat_fun, adjacency, n_jobs, seed, max_step, spatial_exclude, step_down_p, t_power, out_type, check_disjoint, buffer_size, verbose)
   [1546](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1546) else:
   [1547](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1547)     exclude = None
-> [1548](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1548) return permutation_cluster_test(
   [1549](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1549)     X,
   [1550](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1550)     threshold=threshold,
   [1551](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1551)     stat_fun=stat_fun,
   [1552](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1552)     tail=tail,
   [1553](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1553)     n_permutations=n_permutations,
   [1554](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1554)     adjacency=adjacency,
   [1555](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1555)     n_jobs=n_jobs,
   [1556](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1556)     seed=seed,
   [1557](https://file+.vscode-resource.vscode-cdn.net/opt/homebrew/anaconda3/envs/LonelyEEG/lib/python3.11/site-packages/mne/stats/cluster_level.py:1557)     max_step=max_step,
[... cut ...]
├☑ qtpy              2.2.0 (PyQt5=5.15.6)
├☑ pyqtgraph         0.13.1
├☑ mne-qt-browser    0.5.0
├☑ ipywidgets        7.7.5
├☑ trame_client      3.4.0
├☑ trame_server      2.17.3
├☑ trame_vtk         2.8.11
├☑ trame_vuetify     2.7.1
└☐ unavailable       ipympl

Ecosystem (optional)
├☑ eeglabio          0.0.2-3
├☑ edfio             0.4.4
├☑ mffpy             0.8.0
├☑ pybv              0.7.5
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo
```