mne-tools/mne-python #12929: mne.combine_evoked with equal weights returns EvokedArray with float nave

Genuster, 2024-10-31:
### Description of the problem

In the documentation `EvokedArray.nave` is typed as int, but that's not what happening with `mne.combined_evoked`. I actually prefer nave being float, but the problem is that it yields long floating points in plots; this wasn't the case in 1.7.1.

### Steps to reproduce

```Python
import mne
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = sample_data_folder / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False).crop(tmax=120)
events = mne.find_events(raw, stim_channel="STI 014")
event_dict = {
    "auditory/left": 1,
    "auditory/right": 2,
    "visual/left": 3,
    "visual/right": 4,
    "face": 5,
    "button": 32,
}
epochs = mne.Epochs(raw, events, tmin=-0.2, tmax=0.5, event_id=event_dict, preload=True)
mne.combine_evoked([epochs['left'].average(), epochs['right'].average()], weights='equal').plot(picks='eeg')
```

### Expected results

nave being rounded in the plot

### Actual results

[image]

### Additional information

Platform             Windows-11-10.0.22631-SP0
Python               3.12.7 (tags/v3.12.7:0b05ead, Oct  1 2024, 03:06:41) [MSC v.1941 64 bit (AMD64)]
Executable           c:\Users\User\Documents\erp\.venv\Scripts\python.exe
CPU                  Intel64 Family 6 Model 158 Stepping 10, GenuineIntel (12 cores)
Memory               31.9 GB

Core
├☑ mne               1.8.0 (latest release)
├☑ numpy             2.1.2 (OpenBLAS 0.3.27 with 12 threads)
├☑ scipy             1.14.1
└☑ matplotlib        3.9.2 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ pandas            2.2.3
└☐ unavailable       sklearn, numba, nibabel, nilearn, dipy, openmeeg, cupy, h5io, h5py

Visualization (optional)
├☑ mne-qt-browser    0.6.3
├☑ ipywidgets        8.1.5
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

agramfort, 2024-10-31:
see https://github.com/mne-tools/mne-python/blob/main/mne/evoked.py#L1615
for why this is happening.
I would rather fix the plots as the numberics are AFAIK statistically
correct

> Message ID: ***@***.***>
>

drammock, 2024-10-31:
> I would rather fix the plots as the numberics are AFAIK statistically correct

agree, simply rounding `nave` on the plot (maybe 2 or 3 decimals?) seems the right fix. Might also be good to replace `=` with `≈`?

PR #12936 (2024-11-01): BUG: Fix evk.nave plotting when it's float
Fixes #12929.

If `evoked.nave` is not int, it will be rounded to 2 decimal places in `evoked.plot()` and `evoked.plot_image()` and `≈` will be plotted instead of `=`.
closed 2024-11-01 by larsoner