mne-tools/mne-python #12930: Hide scrollbars in browser plots in docs

cbrnr, 2024-10-31:
This is a follow-up to #10481. Our docs contain static screenshots of `raw.plot()`, and although most UI elements have been removed, the scrollbars are still visible, e.g. [here](https://mne.tools/stable/auto_tutorials/epochs/40_autogenerate_metadata.html#tut-autogenerate-metadata-ern):

[image]

Even though they do not consume much space, I think they are confusing because they give the impression that the plot is actually interactive (scrollable). Therefore, I suggest to remove scrollbars from these plots in our docs.

Note that a solution via an environment variable (or similar) would be ideal, because the example code should still contain simple calls of `raw.plot()` without any extra arguments to hide the scrollbars (such an argument already exists, but I would not want to include `show_scrollbars=False` in every single example/tutorial which calls this function). Users will want interactive elements when running the examples, but I think we shouldn't display them.

drammock, 2024-10-31:
IIRC this would go away if the doc builds used MPL backend instead of Qt-browser backend, but we'd have to confirm that none of the plots in our docs are meant to showcase Qt-browser-specific capabililties (as feature parity has drifted recently).

Another possible approach would be to make the `_MNEQtBrowserScraper` class somehow suppress the scrollbars before scraping the image, not sure how feasible that is.

Another possibility is to check for `_MNE_BUILDING_DOCS` env var, and if `True`, suppress the scrollbars

cbrnr, 2024-10-31:
Switching to Matplotlib sounds like the least effort, but since Qt is now the default backend, adapting the scraper would probably be the most uncontroversial option.

drammock, 2024-10-31:
> adapting the scraper would probably be the most uncontroversial option.

I'm OK with any of the 3 possibilities I mentioned before, but I agree that adapting the scraper is probably the most sensible.  Although the 3rd option (check in qt-browser code for the `MNE_BUILDING_DOCS` env var, and if found, hide scrollbars) is also fine with me and possibly easier?

closed 2024-11-04 by larsoner