mne-tools/mne-python #12944: Installation instructions: add IDE setup to installer page

SophieHerbst, 2024-11-05:
### Proposed documentation enhancement

When trying to instruct a new student to install MNE via the installers, we got stuck on the instruction pages, trying to find the steps for setting up spyder. I think it would be useful to have some setup/ getting started info (or a link) on the info page for the installers, as newbies do not go into the pip/conda section where some info can be found. 
I thought this had been there at some point, but I couldn't find it.
Also, it seems the info about VS-code setup disappeared in the doc in version 1.8

hoechenberger, 2024-11-05:
I think this is all due to some re-structuring of the docs that happened quite recently. Might need to re-think this if it's not easy to use

larsoner, 2024-11-05:
Indeed I was hoping this link toward the bottom was discoverable enough:

[image]

But maybe keeping a second-level heading or something would be better. Maybe a `toctree` and a separate page so it shows up in the right sidebar?

SophieHerbst, 2024-11-05:
I think the confusion was that the link takes you to the somewhere inside the description of the conda installation, even though the corresponding text talks about both. I think it would be clearer to make the IDE a separate page and have different sections about the integration with different types of installations.

larsoner, 2024-11-05:
Makes sense!

I think by splitting it up into separate pages we could actually keep the nice structure we have now but *also* have a `.. toctree:: hidden` that makes it show up in the right sidebar. I'll see if I can cook something up

hoechenberger, 2024-11-05:
> But maybe keeping a second-level heading or something would be better. Maybe a `toctree` and a separate page so it shows up in the right sidebar?

Actually ideally the "first steps" would somehow pop up after the user has clicked the download button

larsoner, 2024-11-05:
> Actually ideally the "first steps" would somehow pop up after the user has clicked the download button

Can you try to implement what you have in mind perhaps in #12947 ?

Currently on `main` and in this PR they are shown from the start:

[image]

It sounds like you are now suggesting to have them start hidden but then un-hide when the download button is clicked? It seems like content that's useful enough to always be shown. Maybe it would be better to highlight it somehow (change the background color / class to `success` or something?) rather than un-hide

hoechenberger, 2024-11-05:
I think what you've been working on in that recent PR is fine 😊

PR #12947 (2024-11-05): DOC: Reflow IDE doc
Closes #12944

@SophieHerbst @hoechenberger can you see if this is better? Changes:

1. Split the IDE section into its own `ides.rst` file and added it to the hidden toctree so now `"IDE integration (VSCode, Spyder, etc.)"` is an entry in the left sidebar

   | main | PR |
   | -- | -- |
   | [image] | [image] |

3. Added a card that pops up when you click "download" that links to that section:

   | before click | after click |
   | -- | -- |
   | [image] | [image] |