mne-tools/mne-python #12949: AttributeError: No mne.io attribute read_raw_ant

Nico-La-Porta, 2024-11-06:
### Description of the problem

MNE version: e.g. 1.9.0.dev90+g2983d2a1914
Operating system: Windows 11

Hi everyone.
I'm facing an issue opening a raw ANT files in .cnt format.
I tried this way

`try:
    raw = mne.io.read_raw_cnt(cnt_data[0], preload=True)
except Exception as err:
    print(err,'\n')
    if 'mne.io.read_raw_ant' in str(err):
        try:
            raw = mne.io.read_raw_ant(cnt_data[0], preload=True)
        except Exception as err:
            print(err)`

and got the following AttributeError:
`AttributeError: No mne.io attribute read_raw_ant`

### Steps to reproduce

```Python
Open ANT .cnt file with mne.io.read_raw_ant()
```

### Expected results

I expect to be able to load the ANT .cnt file or, at least, get another error with respect to AttributeError

### Actual results

AttributeError: No mne.io attribute read_raw_ant

### Additional information

No additional information needed

larsoner, 2024-11-06:
`read_raw_ant` should come out as part of 1.9 which hasn't been released yet. In the meantime you can install the dev version if you want to try it out before release!

larsoner, 2024-11-06:
... oh you apparently have 1.9 dev installed. Are you sure it's up to date? If you do
```
import mne
print(mne.__file__)
print(mne.__version__)
```
does it point to the file and version you expect?

Nico-La-Porta, 2024-11-07:
Hi @larsoner, thanks for answering.
Actually, it does not point to the 1.9 dev version but to the 1.8.0 version, despite I am sure to have installed only the 1.9 dev version (I created a new venv, installing only with the 1.9 dev version).

Here is an extract of my pip list:
> mne                       1.9.0.dev91+g649857aacb2
> mne-icalabel              0.7.0

I double-checked in the venv local folder and there is no trace of mne 1.8.0

larsoner, 2024-11-07:
I would do stuff like `which python`, `which pip`, and `pip --version`, I suspect the `pip` you are using maybe isn't the correct one, or an env isn't being activated properly. Or maybe you installed a dev version before `read_raw_ant` was added?

closed 2024-11-08 by Nico-La-Porta