mne-tools/mne-python #12956: TFR Analysis and Cluster-Based Permutation with 1 sample and two conditions

biagioscalingipsy, 2024-11-11:
MNE version: 1.8.0
operating system: macOS Sonoma 14.3.1
Hi everyone,
I’m a beginner with MNE and I’ve recently started working on time-frequency analysis with my EEG data. I’m analyzing data from a within-subjects design with 36 participants and two conditions (metaphor and literal) that I want to compare. The event of interest spans from 0 ms to 2000 ms, so I set up my epochs as follows:

`epochs_images = mne.Epochs(raw, events, event_id=event_id, tmin=-1.000, tmax=2.500, baseline=(-0.750, -0.250), preload=True)`

After that, I ran a time-frequency analysis using Morlet wavelets. Starting from prior literature, I tried two approaches for the `n_cycles` parameter: firstly using a fixed value of 3 cycles, secondly half the frequency:

```
freqs_low = np.arange(5, 41, 2)
n_cycles = 3

freqs_low = np.arange(5, 41, 2)
n_cycles = freqs_low / 2
```

I then ran the TFR analysis like this:
`power_morlet = mne.time_frequency.tfr_morlet(epochs_condition, freqs=freqs_low, n_cycles=n_cycles, use_fft=True, return_itc=False, decim=1, n_jobs=-1)`

Next, I organized my data for the two conditions into separate lists and cropped the time range of interest `tmin, tmax = -0.7, 2.0`. Data were then transposed as `(0, 2, 3, 1)` and an adjacency matrix was created:
```
sensor_adjacency, ch_names = mne.channels.read_ch_adjacency("easycapM1")
common_channels = [ch for ch in metaphor_info.info['ch_names'] if ch in ch_names]
common_indices = [ch_names.index(ch) for ch in common_channels]

# Aggiorna sensor_adjacency per i canali comuni
filtered_sensor_adjacency = sensor_adjacency[common_indices][:, common_indices]

# Combina la matrice di adiacenza con quella temporale e frequenziale
combined_adjacency = combine_adjacency(len(metaphor_info.freqs), len(metaphor_info.times),filtered_sensor_adjacency)
```
[image]

I tried a cluster permutation test with the following code:

```
X = metaphor_data_transposed - literal_data_transposed
n_observations = len(metaphor_data_transposed)
pval = 0.05
df = n_observations - 1
thresh = scipy.stats.t.ppf(1 - pval / 2, df)

n_permutations = 1024

T_obs, clusters, cluster_p_values, H0 = permutation_cluster_1samp_test(
    X,
    n_permutations=n_permutations,
    threshold=thresh,
    tail=0,
    adjacency=combined_adjacency,
    out_type="mask",
    verbose=True,
    n_jobs=-1,
    seed=42
)
```
The tests returned 11 clusters, but none of them were significant:

```
Using a threshold of 2.030108
stat_fun(H1): min=-3.882895189270287 max=2.4152846969668698
Running initial clustering …
Found 11 clusters
```
```
array([1.        , 1.        , 1.        , 0.94628906, 1.        ,
       1.        , 1.        , 1.        , 1.        , 1.        ,
       1.        ])
```

 I think it is quite strange that there are no significant differences between the conditions, so I’m wondering if I might be doing something incorrectly.
Any guidance on my approach or suggestions for troubleshooting would be greatly appreciated. Thanks in advance for your help!

scott-huberty, 2024-11-16:
Hi @biagioscalingipsy - Please post your usage related questions on the [MNE user forum](https://mne.discourse.group). Github issues is for bug reports / feature requests, etc.

scott-huberty, 2024-11-22:
This [is being discussed](https://mne.discourse.group/t/tfr-analysis-and-cluster-based-permutation-with-1-sample-and-two-conditions/10500/8) on the user forum. Should we Close?

closed 2024-11-22 by larsoner