mne-tools/mne-python #12959: `.add_reference_channels` sends unexpected log about locations

sappelhoff, 2024-11-12:
This log message is unexpected 👇 ...it shows up even when no locations have been in the raw object to begin with.

https://github.com/mne-tools/mne-python/blob/9832d4b7a9935898aa6ff6ddd0808e8b5aa3d7aa/mne/_fiff/reference.py#L303-L308

MWE 👇 

```python
# %%
# Imports
import mne

# %%
# Get some test data
sample = mne.datasets.sample.data_path()
raw = mne.io.read_raw(sample / "MEG" / "sample" / "sample_audvis_raw.fif")
raw.pick("eeg")  # for simplicity
raw.set_montage(None)  # remove all locations
raw.load_data()

# %%
# This logs:
# Location for this channel is unknown;
# consider calling set_montage() after adding new reference channels if needed.
# Applying a montage will only set locations of channels that exist at the time it is applied.
raw.add_reference_channels("ZZZ")

# ... I find that unexpected / unneeded: This log should only occur for raw objects that
# already have SOME locations. NOT for raw objects that do not have any locations anyhow.
# %%
```

☝️ IMHO we should only send this log if *some* locations were already applied at some point.

larsoner, 2024-11-12:
Makes sense to me!

sappelhoff, 2024-11-20:
I actually find the code a bit confusing and do not really know what to remove:

```Python
    # only do this if we actually have digitisation points
    if inst.info.get("dig", None) is not None:
        # "zeroth" EEG electrode dig points is reference
        ref_dig_loc = [
            dl
            for dl in inst.info["dig"]
            if (dl["kind"] == FIFF.FIFFV_POINT_EEG and dl["ident"] == 0)
        ]
        if len(ref_channels) > 1 or len(ref_dig_loc) != len(ref_channels):
            ref_dig_array = np.full(12, np.nan)
            warn("The locations of multiple reference channels are ignored.")
        else:  # n_ref_channels == 1 and a single ref digitization exists
            ref_dig_array = np.concatenate(
                (ref_dig_loc[0]["r"], ref_dig_loc[0]["r"], np.zeros(6))
            )
            # Replace the (possibly new) Ref location for each channel
            for idx in pick_types(inst.info, meg=False, eeg=True, exclude=[]):
                inst.info["chs"][idx]["loc"][3:6] = ref_dig_loc[0]["r"]
    else:
        # Ideally we'd fall back on getting the location from a montage, but
        # locations for non-present channels aren't stored, so location is
        # unknown. Users can call set_montage() again if needed.
        ref_dig_array = np.full(12, np.nan)
        logger.info(
            "Location for this channel is unknown; consider calling "
            "set_montage() after adding new reference channels if needed. "
            "Applying a montage will only set locations of channels that "
            "exist at the time it is applied."
        )

```

☝️ does this mean that sometimes we CAN set a channel location for an added reference channel?

Else I would simply suggest to entirely remove the `logger.info` call from the `else` clause above.

larsoner, 2024-11-22:
> ☝️ does this mean that sometimes we CAN set a channel location for an added reference channel?

Yes I think so, assuming it is the one ref channel present in the dataset and there is an existing dig point for it

closed 2024-11-22 by larsoner