mne-tools/mne-python #12963: Error when anonymizing via `meas_info.anonymize()`

berkgercek, 2024-11-13:
### Description of the problem

Attempting to use the command `mne anonymize -f ./myfile.fif` from version 1.8.0 onwards results in one of two errors. 

The first possible error is during a call to `mne._fiff.meas_info._add_timedelta_to_stamp`, which results in the error:

```
Traceback (most recent call last):
  File "/tmp/gercek_li_venv/bin/mne", line 8, in <module>
    sys.exit(main())
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/commands/utils.py", line 109, in main
    cmd.run()
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/commands/mne_anonymize.py", line 122, in run
    mne_anonymize(fname, out_fname, keep_his, daysback, overwrite)
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/commands/mne_anonymize.py", line 47, in mne_anonymize
    raw.anonymize(daysback=daysback, keep_his=keep_his)
  File "<decorator-gen-26>", line 12, in anonymize
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/_fiff/meas_info.py", line 793, in anonymize
    anonymize_info(info, daysback=daysback, keep_his=keep_his, verbose=verbose)
  File "<decorator-gen-34>", line 12, in anonymize_info
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/_fiff/meas_info.py", line 3529, in anonymize_info
    _add_timedelta_to_stamp(hi["meas_date"], -delta_t)
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/_fiff/meas_info.py", line 3386, in _add_timedelta_to_stamp
    meas_date_stamp = _dt_to_stamp(_stamp_to_dt(meas_date_stamp) + delta_t)
  File "/tmp/gercek_li_venv/lib/python3.10/site-packages/mne/utils/numerics.py", line 1012, in _stamp_to_dt
    stamp = [int(s) for s in utc_stamp]
TypeError: 'datetime.datetime' object is not iterable
```

This happens because it seems the `datetime.datetime` object is not iterable, and my FIF file (MEGIN Triux Neo) only has a single DT object as the helium measurement date.

The second possible error is if `raw.info["meas_date"]` has been set to `None` prior to the anonymize script being run, but the helium info `meas_date` is still a datetime value. I believe `mne-bids-pipeline` does something to this effect when it runs the pipeline step `preprocessing/data_quality`. This results in an error when attempting to set the measurement date to None:

```
helium_info['meas_date'] must be an instance of datetime, got <class 'NoneType'> instead.
```

I don't have the stack trace on hand for this one, but it occurs during at [this line, when the code attempts to set the `HeliumInfo` attribute to None](https://github.com/mne-tools/mne-python/blob/c3c1f2934f23bbb8409094c06576f7a470357abe/mne/_fiff/meas_info.py#L3525) and `None` isn't an accepted value for the `ValidatedDict` subclass `HeliumInfo`.

### Steps to reproduce

```Python
# Error version 1:
$ mne anonymize -f ./myfile.fif 

# Error version 2:
$ mne_bids_pipeline --steps preprocessing
```

### Expected results

It should be possible, given that MNE supports anonymization, to set the field in question to `None`. I'm less sure about why the iterable error is happening.

### Actual results

See above

### Additional information

I've tried this with MNE versions 1.7.1, 1.8.0, and 1.9.0-dev. Both categories of error only occur with 1.8 and 1.9.

The .fif file in question is from a MEGIN Triux Neo, I can provide an empty-room file that produces the error on request, but it's a pain to upload.

larsoner, 2024-11-13:
Indeed this looks like an error, I should be able to look tomorrow!

berkgercek, 2024-11-14:
Thanks Eric!

PR #12967 (2024-11-14): BUG: Fix bug with anonymization
Closes #12963

Within-cycle fix so no need for changelog entry.

Our anonymization tests were a bit lacking so changes ended up being a bit bigger than I expected.
closed 2024-11-14 by larsoner