mne-tools/mne-python #12977: Repairing artifacts with regression - Using the wrong regression coefficients

JacPhe, 2024-11-20:
### Proposed documentation enhancement

In the aforementioned tutorial, under "Subtract the evoked response from the epoch data before regression":

I think that
epochs_clean_sub = model_plain.apply(epochs)
is incorrect, as this applies the regression coefficients from model_plain, which was trained on the default epochs, rather than epochs.subtract_evoked().

I think it should be
epochs_clean_sub = model_sub.apply(epochs)
instead. Otherwise, model_sub appears to be redundant.

scott-huberty, 2024-11-20:
Huh, I think you are right @JacPhe . Would you be up for a PR to fix this?

Also just to be safe, CC'ing @wmvanvliet as the author of that tutorial... On line 152 shouldn't `model_plan.apply(...)` actually be `model_sub.apply(...)`?

https://github.com/mne-tools/mne-python/blob/a99aa9f798d878ba144b3f6f3127311841d5d2b5/tutorials/preprocessing/35_artifact_correction_regression.py#L143-L154

JacPhe, 2024-11-20:
I've sent a pull request just in case, hopefully that helps!

PR #12978 (2024-11-22): Changed epochs_clean_sub = model_plain.apply(epochs) to epochs_clean_sub = model_sub.apply(epochs), making use of the regression coefficients from model_sub. #12977
Fixes #12977  

Originally, the documentation has epochs_clean_sub = model_plain.apply(epochs), which uses model_plain. This contains regression coefficients for EOG, with some bleed in from adjacent EEG sensors. I believe that it should be using epochs_clean_sub = model_sub.apply(epochs), as this makes use of model_sub, which has the regression coefficients for EOG, with a subtraction of epochs.evoked. I think this makes sense bec [... cut]
closed 2024-11-22 by wmvanvliet