mne-tools/mne-python #12979: 'mode' for mne.sensitivity_map not explained

olafhauk, 2024-11-21:
### Proposed documentation enhancement

The current documentation for the option 'mode' of mne.sensitivity_maps refers to a "manual" (presumbably the C-version of MNE), but there is no link (see below). I think the different choices should be explained in the documention of MNE-Python, without reference to a manual.

mode: str
The type of sensitivity map computed. See manual. Should be 'free', 'fixed', 'ratio', 'radiality', 'angle', 'remaining', or 'dampening' corresponding to the argument --map 1, 2, 3, 4, 5, 6, 7 of the command mne_sensitivity_map.

larsoner, 2024-11-22:
@olafhauk would you be up for taking the text from here and adding it?
```
$ mne_sensitivity_map --help
usage: mne_sensitivity_map [options]
Compute MEG and EEG sensitivity maps over the cortex.
	--fwd name        The input file.
	--proj name       Load a projection operator from this file
	--eeg             Include EEG instead of MEG.
	--mag             Include magnetometers instead of gradiometers.
	--map number      Select the quantity to compute:
	                  1 : Maximum obtainable signal among all orientations
	                  2 : Signal from sources normal to the cortex
	                  3 : Ratio of signal from sources normal to the cortex and max
	                  4 : Ratio of signal from sources normal to the cortex and max
	                  5 : Subspace angle with the noise subspace
	                  6 : Fraction of signal remaining after the projection
	                  7 : Fraction of signal dampening due to the projection
```

NatalieNegussie97, 2025-08-15:
Hi :)
I'd like to work on this. I'm new to contributing to MNE, so please let me know if there are any preferred guidelines or wording. I'll add a clear explanation of the mode options to the API docstring of mne.sensitivity_map (and link from the docs if needed). If there’s any preferred wording, please let me know.