mne-tools/mne-python #12980: RFC:  splitting simulation code into separate package

drammock, 2024-11-21:
Hi @mne-tools/mne-python-steering-committee 

@ctrltz and colleagues have been doing some work on simulating MNE objects (above and beyond what we currently support), in the new https://github.com/ctrltz/meegsim package.  They have expressed interest in incorporating their work into MNE-Python, and after a few discussions with @larsoner @agramfort and @britta-wstnr we've settled on a tentative plan to "spin off" our existing simulation code into a new package (`mne-simulation` or so) similar to what was done for connectivity a few years ago, and then give @ctrltz and colleagues write-access to that repo so they can add their code and also have a much smaller/faster test/CI suite.  Existing simulation code would be deprecated in MNE-Python release 1.9 and removed in 1.10.

This issue is your chance to weigh in on whether you think it best to:

1. follow the above plan
2. keep existing simulation code in MNE-Python, and add new code into the main repo
3. keep existing simulation code in MNE-Python, and don't add the new MEEGsim functionality
4. something else (?)

drammock, 2024-11-21:
one possible downside to the "spinoff" approach that I've just thought of, is the risk of losing git history for the existing code.  I think there's probably a way to preserve it, via something like:

- fresh-clone MNE-Python repo
- filter + rewrite history to only include the simulation files
- make a commit that fixes all the imports (e.g., changes `from .forward import whatever` to `from mne.forward import whatever`)
- add in all the other packaging-related scaffolding

ctrltz, 2024-11-21:
Thank you, @drammock! Tagging @minajamshidi and @astudenova to keep them in the loop!

hoechenberger, 2024-11-21:
Go for it 👍

cbrnr, 2024-11-21:
Also 👍 for spin off!

britta-wstnr, 2024-11-21:
+1 from me, too. Really excited about the cool options that @ctrltz, @minajamshidi, and @astudenova support in MEEGsim! 😍

adam2392, 2024-11-21:
+1 supporting simulation would help so much research in this space.

agramfort, 2024-11-23:
+1 too.

We could even start thinking into adding ML based generative models. There is a bunch of cool works in this space eg

https://www.biorxiv.org/content/10.1101/2023.08.23.554148v1
https://github.com/LiuyinYang1101/EEGDiffWave

drammock, 2024-11-25:
OK thanks everyone for weighing in. I'll start working on the deprecation + spin-off this week. I'll leave this open, but convert the title to be normative, so we can close it when the spin-off is complete.