mne-tools/mne-python #12982: Bug when read raw cnt file

ZiyiTsang, 2024-11-22:
### Description of the problem

Hi who it may concern, I am writing the code to read cnt when it occur the error as follow:

**Error**
File /mnt/nvme2n1/(xxxxxxxxxx)/lib/python3.10/site-packages/mne/io/cnt/cnt.py:340, in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, header)
    338 # See: PR #12393
    339 annotations = _read_annotations_cnt(input_fname, data_format="int16")
--> 340 if annotations.onset[-1] * sfreq > n_samples:
    341     n_bytes = 4
    342     n_samples = n_samples_header

IndexError: index -1 is out of bounds for axis 0 with size 0

### Steps to reproduce

```Python
data = mne.io.read_raw_cnt('1_1.cnt', preload=True)

(then orrur happen...)
```

### Link to data

This is a file in  [SEED-GER dataset](https://bcmi.sjtu.edu.cn/~seed/seed-GER.html). I can send you the file email for debug purpose only.

### Expected results

Expected run perfectly

### Actual results

[image]
**Traceback**
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[15], line 3
      1 with warnings.catch_warnings():
      2     warnings.simplefilter("ignore")
----> 3     data = mne.io.read_raw_cnt('/mnt/xxxxxxxxxxx/SEED_GER/German/01-EEG-raw/1_1.cnt', preload=True)

File /mnt/xxxxxxxxxxx/python3.10/site-packages/mne/io/cnt/cnt.py:248, in read_raw_cnt(input_fname, eog, misc, ecg, emg, data_format, date_format, header, preload, verbose)
    163 @fill_doc
    164 def read_raw_cnt(
    165     input_fname,
   (...)
    175     verbose=None,
    176 ) -> "RawCNT":
    177     """Read CNT data as raw object.
    178 
    179     .. Note::
   (...)
    246     .. versionadded:: 0.12
    247     """
--> 248     return RawCNT(
    249         input_fname,
    250         eog=eog,
    251         misc=misc,
    252         ecg=ecg,
    253         emg=emg,
    254         data_format=data_format,
    255         date_format=date_format,
    256         header=header,
    257         preload=preload,
    258         verbose=verbose,
    259     )

File /mnt/xxxxxxxxxxxx/python3.10/site-packages/mne/io/cnt/cnt.py:531, in RawCNT.__init__(self, input_fname, eog, misc, ecg, emg, data_format, date_format, header, preload, verbose)
    528     _date_format = "%m/%d/%y %H:%M:%S"
    530 input_fname = path.abspath(input_fname)
--> 531 info, cnt_info = _get_cnt_info(
    532     input_fname, eog, ecg, emg, misc, data_format, _date_format, header
    533 )
    534 last_samps = [cnt_info["n_samples"] - 1]
    535 super().__init__(
    536     info,
    537     preload,
   (...)
    542     verbose=verbose,
    543 )

File /mnt/xxxxxxxxxxxxxxxx/python3.10/site-packages/mne/io/cnt/cnt.py:340, in _get_cnt_info(input_fname, eog, ecg, emg, misc, data_format, date_format, header)
    338 # See: PR #12393
    339 annotations = _read_annotations_cnt(input_fname, data_format="int16")
--> 340 if annotations.onset[-1] * sfreq > n_samples:
    341     n_bytes = 4
    342     n_samples = n_samples_header

IndexError: index -1 is out of bounds for axis 0 with size 0

### Additional information

Platform             Linux-5.15.0-124-generic-x86_64-with-glibc2.31
Python               3.10.15 (main, Oct  3 2024, 07:27:34) [GCC 11.2.0]
Executable           /mnt/nvme2n1/zengziyi/anaconda/envs/LGU/bin/python
CPU                  x86_64 (112 cores)
Memory               1007.5 GB

Core
├☑ mne               1.8.0 (latest release)
├☑ numpy             1.26.4 (MKL 2023.1-Product with 56 threads)
├☑ scipy             1.13.1
└☑ matplotlib        3.9.2 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.5.2
├☑ pandas            2.2.2
├☑ h5io              0.2.4
└☐ unavailable       numba, nibabel, nilearn, dipy, openmeeg, cupy, h5py

Visualization (optional)
├☑ ipywidgets        8.1.5
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
├☑ mne-icalabel      0.7.0
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

larsoner, 2024-11-22:
Indeed this looks like it's probably a bug where we assume that there is at least one annotation, but there are none. 

If you change

https://github.com/mne-tools/mne-python/blob/f5d043a34e3dda3c83401a68d5722c1c664dc925/mne/io/cnt/cnt.py#L340

To
```
if len(annotations) and annotations.onset[-1] * sfreq > n_samples:
```
does it work? If so, PR welcome for this quick fix!

ZiyiTsang, 2024-11-23:
Ok, trying to working on it!

ZiyiTsang, 2024-11-23:
Yes, it work for me. I have create PR #12986 to quick fix this.
This is my first time creating PR. If anything wrong, kindly inform me. Thanks.

ZiyiTsang, 2024-11-26:
As PR complete, I close this issue. Thanks!

closed 2024-11-26 by ZiyiTsang