mne-tools/mne-python #12987: Code for projecting 3D spherical coordinates into 2D plane coordinates

sdsafsdfsaf, 2024-11-25:
``` python
locs3d -= sphere[:3]
 # use spherical (theta, pol) as (r, theta) for polar->cartesian
cart_coords = _cart_to_sph(locs3d)
out = _pol_to_cart(cart_coords[:, 1:][:, ::-1])
# scale from radians to mm
out *= cart_coords[:, [0]] / (np.pi / 2.0)
out += sphere[:2]
```
``` python
def _pol_to_cart(pol):
    """Transform polar coordinates to cartesian."""
    out = np.empty((len(pol), 2))
    if pol.shape[1] == 2:  # phi, theta
        out[:, 0] = pol[:, 0] * np.cos(pol[:, 1])
        out[:, 1] = pol[:, 0] * np.sin(pol[:, 1])
    else:  # radial distance, theta, phi
        d = pol[:, 0] * np.sin(pol[:, 2])
        out[:, 0] = d * np.cos(pol[:, 1])
        out[:, 1] = d * np.sin(pol[:, 1])
    return out
```
I cannot understand how the calculation formula for projecting condition 1 from 3D spherical coordinates to a two-dimensional plane is derived. Can you provide some references?

sdsafsdfsaf, 2024-11-26:
I have also inquired about chatgpt, but after careful comparison, I found that the result was incorrect

sdsafsdfsaf, 2024-11-26:
> Hi! I quickly copied your question into ChatGPT (model o1-preview)
> 
> See the response here:
> 
> https://chatgpt.com/share/67457e4f-3be0-800e-9737-293f51737d8d
> 
> (If the page doesn't load on the first try, click the link again)

Thank you, I have also inquired about chatgpt before, but after careful comparison, I found that the result was incorrect. This projection does not seem to simply take the original 3D coordinates of x and y, but rather performs a certain projection transformation on spherical coordinates, making the newly obtained x and y different from the original.

hoechenberger, 2024-11-26:
> This projection does not seem to simply take the original 3D coordinates of x and y, but rather performs a certain projection transformation on spherical coordinates,

In the first step, the origin is shifted:

> ```python
> locs3d -= sphere[:3]
> ```

And after that, MNE converts the cartesian XYZ coordinates to polar corrdinates, and then projects those to XY plane and scales the result. Maybe I'm misunderstanding your question 🤔

sdsafsdfsaf, 2024-11-27:
> > This projection does not seem to simply take the original 3D coordinates of x and y, but rather performs a certain projection transformation on spherical coordinates,
> 
> In the first step, the origin is shifted:
> 
> > ```python
> > locs3d -= sphere[:3]
> > ```
> 
> And after that, MNE converts the cartesian XYZ coordinates to polar corrdinates, and then projects those to the cartesian XY plane and scales the result. Maybe I'm misunderstanding your question 🤔

Thank you very much, I made a mistake with the theta angle and mistakenly identified its complementary angle as the theta angle. I understand the calculation process of condition 2, but I still don't understand why the calculation of condition 1 is like this? And why is the scaling of x and y done in this way?

``` python
out *= cart_coords[:, [0]] / (np.pi / 2.0)
```
It's a bit strange that we have obtained the three-dimensional coordinates xyz, so why don't we just take xy, but instead need to transform it like this?

hoechenberger, 2024-11-27:
There are numerous way to project points on a sphere onto a 2d plane. We simply chose one that we thought was most appropriate for the kind of data we want to visualize. Think of all the different projections of the globe onto a 2d map... none of them is "correct", they all have advantages and disadvantages.

I believe what we're using here is a form of [azimuthal equidistant projection](https://en.wikipedia.org/wiki/Azimuthal_equidistant_projection?wprov=sfti1).

sappelhoff, 2024-11-27:
> Think of all the different projections of the globe onto a 2d map... none of them is "correct", they all have advantages and disadvantages.

agreed :)

I personally always found https://en.wikipedia.org/wiki/Stereographic_projection to be most easy to understand.

hoechenberger, 2024-11-27:
@sdsafsdfsaf I hope we could help you a bit. If you have any further questions, please refer to our forum, which is more suitable for this kind of discussion than our GitHub issue tracker. Thanks!