mne-tools/mne-python #12993: Inconsistent API? `mne.channels.read_polhemus_fastscan`

sappelhoff, 2024-11-27:
Why is [`mne.channels.read_polhemus_fastscan`](https://mne.tools/stable/generated/mne.channels.read_polhemus_fastscan.html) not called `mne.channels.read_dig_polhemus_fastscan`, with the `read_dig` prefix that all other functions of this family have?

- https://mne.tools/stable/generated/mne.channels.read_polhemus_fastscan.html
- ..........................................................................................................☝️ 👇 
- https://mne.tools/stable/generated/mne.channels.read_dig_polhemus_isotrak.html
- https://mne.tools/stable/generated/mne.channels.read_dig_captrak.html
- https://mne.tools/stable/generated/mne.channels.read_dig_dat.html
- https://mne.tools/stable/generated/mne.channels.read_dig_egi.html
- https://mne.tools/stable/generated/mne.channels.read_dig_localite.html
- https://mne.tools/stable/generated/mne.channels.read_dig_fif.html

I suggest we deprecate `mne.channels.read_polhemus_fastscan` and reintroduce it as `mne.channels.read_dig_polhemus_fastscan`

larsoner, 2024-12-02:
> Why is mne.channels.read_polhemus_fastscan not called mne.channels.read_dig_polhemus_fastscan, with the read_dig prefix that all other functions of this family have?

I think the motivation is that it does not return a `montage` instance like all of the others do, but rather just an `ndarray`

sappelhoff, 2024-12-02:
> I think the motivation is that it does not return a montage instance like all of the others do, but rather just an ndarray

Good point, I didn't see that.

And is there a reason it returns the dig points, rather than a montage? I assume users would turn this into a montage themselves, anyhow? 🤔

larsoner, 2024-12-02:
I *think* the only information is the 3D positions. Other montage formats have other stuff I think, like channel names. Hard to make a usable DigMontage without that info

sappelhoff, 2024-12-02:
From the source code, that seems to be the case indeed. That's enough of an explanation for me to close this, thanks Eric!

closed 2024-12-02 by sappelhoff