mne-tools/mne-python #12994: Allow for roundtrip between `DigMontage.save()` and `mne.channels.read_dig_fif`

sappelhoff, 2024-11-27:
### Describe the new feature or enhancement

I would like to save a `DigMontage` object, and later be able to get the full object back via `mne.channels.read_dig_fif` 

### Describe your proposed implementation

Currently, `ch_names` are not saved, and I don't know which parts of FIF would need to change to allow for this.

### Describe possible alternatives

We could offer a `format` parameter in `DigMontage.save`, which can be `fif` by default (the current behavior), but could also be set to some other format, which would then allow a roundtrip reading (e.g., CapTrak: https://mne.tools/stable/generated/mne.channels.read_dig_captrak.html#mne.channels.read_dig_captrak)

sappelhoff, 2024-11-30:
Actually, one could already set a montage to an info object, and save the info, and then use the get method to get the montage back. That would preserve channel names ... But it's a bit unclean to recommend doing that, as an info object also needs to be set with a sampling frequency, which may not be applicable if you really only have a montage.

An additional point is that for most fif files we save, we recommend a suffix like raw, ave, or info. However for a dig montage, there is no such recommendation.

larsoner, 2024-12-02:
> Currently, ch_names are not saved, and I don't know which parts of FIF would need to change to allow for this.

I don't think any parts of the FIF spec would need to change, in principle I think you can use `write_name_list_sanitized` and the corresponding read function

https://github.com/mne-tools/mne-python/blob/096243fe43c936587190a8e9e8e86b155446800e/mne/_fiff/write.py#L152

Similar to how we do here for bads

https://github.com/mne-tools/mne-python/blob/096243fe43c936587190a8e9e8e86b155446800e/mne/_fiff/meas_info.py#L2092

Because you are writing a CH_NAME_LIST, but instead of `"bads"` you would have `"ch_names"`.

closed 2024-12-04 by larsoner