mne-tools/mne-python #13016: Release 1.9

larsoner, 2024-12-09:
We're due to release 1.9 this week. I looked through open issues and PRs and marked what I see (and am willing to champion/fix) to get in within the next few days. Anything else that was marked for 1.9 I didn't consider a blocker so marked as 1.10, but others feel free to take a look and re-milestone stuff back for 1.9 if you see things that should block.

I think we had targeted Thurs 12/12 for release, which seems okay to me based on the current list.

sappelhoff, 2025-01-02:
@larsoner is this a typo (missing the `Version` prefix)?

https://mne.tools/stable/development/whats_new.html

[image]

larsoner, 2025-01-07:
Thanks for fixing that @sappelhoff !

closed 2025-01-07 by larsoner