mne-tools/mne-python #13023: Possible bug in computation of multitaper TFR power

tsbinns, 2024-12-12:
### Description of the problem

When working through some bugs with plotting TFR multitaper data in #12910 I was looking over the `_time_frequency_loop` code for converting complex multitaper coefficients into real-valued power.

Here the timeseries data, `X`, and windows, `Ws`, for each taper, `W`, are convolved to give the time-frequency transformed data, `coefs`:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/tfr.py#L709-L713

When power is requested, the complex coefficients for each taper are converted to power:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/tfr.py#L719-L723

... and summed together:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/tfr.py#L740

But I don't see at any point a weighting of the coefficients according to the taper weights. There is at the end a normalisation according to the number of tapers:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/tfr.py#L752-L754

... but that just means each taper is treated as having an equal weighting.

I also didn't notice anywhere upstream of the TFR computation where the taper weights were already applied.

This could be a misunderstanding on my part, but at least based on how PSD computation is handled, taper weights are applied when converting complex coefficients to power:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/multitaper.py#L187-L206

<br>

@larsoner do you have any ideas?

### Steps to reproduce

```Python
N/A
```

### Expected results

N/A

### Actual results

N/A

### Additional information

N/A

larsoner, 2024-12-16:
Yeah just the fact that we call `_make_dpss` without the `return_weights=True` suggests that they aren't/can't be used. It seems like they should be applied to the `Ws` and then normalized properly at the end.

tsbinns, 2024-12-16:
Okay, so #12910 makes the weights available in the class which could be used. Is it worth just fixing this there, or open a new PR? Don't want to overburden with a big PR review.

larsoner, 2024-12-16:
We're cutting a release now-ish. After that let's merge #12910 then a separate PR (which you could start now based on #12910 if you don't mind some rebase/merge mess) let's fix the computation.

closed 2025-01-19 by larsoner