mne-tools/mne-python #13026: Proposal to Add Upper Bounds for Python Version Compatibility in `pyproject.toml`

hoechenberger, 2024-12-13:
I recently encountered an issue while attempting to set up MNE-Python in an environment with a recent Python version (specifically Python 3.13). This is a recurring problem that I've experienced before, but I’ve decided to report it now after finally pinpointing the root cause. For reference, the sam eissue has been reported to the uv developers here: [astral-sh/uv#9864](https://github.com/astral-sh/uv/issues/9864), to hopefully get some input from package management experts as well.

### Issue

When installing MNE in an environment with Python 3.13, `pip` (and `uv`) resolve to an outdated version of MNE (`mne==1.5.1` instead of `mne==1.8.0`) if I specify the `mne[full]` extra. Notably, installing `mne` without the `[full]` extra resolves correctly to version 1.8.0. I suspect this discrepancy arises due to metadata constraints in some of our dependencies.

This highlights a broader concern: we do not currently test or guarantee compatibility of older MNE versions with recent Python releases. While our CI system ensures compatibility for the latest stable and development versions of MNE against supported Python versions, it does not cover older MNE releases. Consequently, an older MNE release e.g. claiming to support Python versions `>=3.`6 may not function correctly with newer Python versions (e.g., Python 3.10 or later).

### Proposed Solution

I propose adding upper bounds to the `requires_python` field in our `pyproject.toml` to reflect the Python versions we have explicitly tested and verified for compatibility. This change would:

- Prevent outdated MNE versions from being installed on Python versions that we cannot guarantee they support.
- Provide a clearer and more predictable user experience by enforcing a "hard stop" when Python versions exceed our tested limits.
- Align user expectations with our actual CI testing scope, ensuring that MNE versions work reliably with the Python versions they claim to support.

### Implementation Details

- For each MNE release, define the range of Python versions supported (e.g., `>=3.6, <3.12` for a release tested only up to Python 3.11).
- When support for a new Python version is confirmed, relax the upper bound and include the updated constraint in the next release.
- Ensure that CI testing includes comprehensive validation for all supported Python versions within the defined range.

### Benefits

- **Improved User Experience:** Users are less likely to encounter unexpected issues caused by installing an outdated MNE version.
- **Clearer Compatibility Expectations:** Version constraints will accurately reflect what we’ve tested and can confidently support.
- **Proactive Error Prevention:** Users attempting to install MNE on unsupported Python versions will receive an immediate and clear error message.

### Considerations

This change would require vigilance when releasing new versions to ensure that the `requires_python` field is updated appropriately. It would also rely on timely updates when support for new Python versions is added. However, I believe these additional steps are justified given the potential for improved reliability and user satisfaction.

cbrnr, 2024-12-13:
I agree that running into this problem can be frustrating, so we should definitely do something to improve the situation. I'm not sure if an upper bound for the Python version is a good idea, because the current best practice seems to recommend no upper boundaries.

Note that currently, there are issues with 3.13 compatibility, which @larsoner is already addressing in #13021.

hoechenberger, 2024-12-13:
> Note that currently, there are issues with 3.13 compatibility

But doesn't this mean that … well, we do **not,** in fact, support Python 3.13 yet, period? How would we prevent the issue I've been running into without adding a boundary?

drammock, 2024-12-13:
> This highlights a broader concern: we do not currently test or guarantee compatibility of older MNE versions with recent Python releases.

On this point I'll also note that our stated policy is that we do not guarantee support for old versions. So incompatibility of MNE v1.5 with (say) NumPy 2.x or Python 3.13 is not something we're on the hook to fix and IMO not something we should really consider prioritizing.

larsoner, 2024-12-13:
@hoechenberger have you looked at other well maintained projects in the Scientific Python ecosystem to see what they do? Often when a big, impactful, generally applicable idea comes up like this those projects have had similar discussions before and made a well thought out decision. And by doing what everyone else does (to the extent possible/reasonable) the community consistency, experiences, and expectations are generally improved. I don't think we should spend too much time discussing this ourselves until we see if we're reinventing the wheel with our discussion here.

> Note that currently, there are issues with 3.13 compatibility, which @larsoner is already addressing in https://github.com/mne-tools/mne-python/pull/13021.

FWIW this isn't an issue with functionality, just minor docstring rendering issues, and therefore a few minor docstring failed tests. I can't think of a time a new Python version has actually broken any *functionality* in MNE-Python, though it will sometimes break unit tests, most often due to new uncaught deprecation warnings. Am I forgetting something here @hoechenberger ?

So the problem here really comes down to (I think): MNE-Python itself works fine for new releases, but *its dependencies in `mne[full]` installation mode* are wide-ranging and include stuff like numba, vtk, Qt, OpenMEEG, etc. that can be slow to update once a new version of Python comes out, causing the problem above.

hoechenberger, 2024-12-13:
> ❯ uv pip install "mne[dev]"

You're mistaken here

This will install stable MNE from PyPI with the `dev` extra. This is **not** our dev version
In fact, we shouldn't even publish the `dev` extra in the first place
There's a new PEP that was recently accepted that solves this issue

cbrnr, 2024-12-13:
Yes, I know. But it doesn't work!

hoechenberger, 2024-12-14:
Ok I think I can (need to!) backtrack here! `uv` 0.5.9 has been released yesterday and changed the default resolution strategy, and it now behaves differently (and as expected, I would say!) for Python 3.13!

First ensure you're on the latest version:
```shell
❯ uv self update
```

Then:
```shell
❯ uv tool install "mne[full]"
Resolved 184 packages in 280ms
  × Failed to download and build `llvmlite==0.43.0`
  ╰─▶ Build backend failed to build wheel through `build_wheel` (exit status: 1)

      [stdout]
      running bdist_wheel
      /Users/richardhochenberger/Library/Caches/uv/builds-v0/.tmpzLAdWT/bin/python
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/llvmlite/0.43.0/8VTFgYgZ65MkutMBVKPSh/src/ffi/build.py
      LLVM version...

      [stderr]
      Traceback (most recent call last):
        File
      "/Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/llvmlite/0.43.0/8VTFgYgZ65MkutMBVKPSh/src/ffi/build.py",
      line 235, in <module>
          main()
          ~~~~^^
        File
      "/Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/llvmlite/0.43.0/8VTFgYgZ65MkutMBVKPSh/src/ffi/build.py",
      line 229, in main
          main_posix('osx', '.dylib')
          ~~~~~~~~~~^^^^^^^^^^^^^^^^^
        File
      "/Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/llvmlite/0.43.0/8VTFgYgZ65MkutMBVKPSh/src/ffi/build.py",
      line 142, in main_posix
          raise RuntimeError(msg) from None
      RuntimeError: Could not find a `llvm-config` binary. There
      are a number of reasons this could occur, please see:
      https://llvmlite.readthedocs.io/en/latest/admin-guide/install.html#using-pip for help.
      error: command '/Users/richardhochenberger/Library/Caches/uv/builds-v0/.tmpzLAdWT/bin/python'
      failed with exit code 1

  help: `llvmlite` (v0.43.0) was included because `mne[full]` (v1.8.0) depends on `numba`
        (v0.60.0) which depends on `llvmlite`
```
It tries to compile `numba` from source, as there are no wheels available for Python 3.13 – which is to be expected, as the latest stable release (0.60.0) does not yet support 3.13.

So instead of an old MNE version, I get a hard stop because I don't have LLVM installed.

Let's try with `mne[dev]`:
```shell
❯ uv tool install "mne[dev]"
Resolved 148 packages in 34ms
  × Failed to download and build `dipy==1.10.0`
  ╰─▶ Build backend failed to build wheel through `build_wheel` (exit status: 1)

      [stdout]
      + meson setup
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src/.mesonpy-az8zmviv
      -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md
      --native-file=/Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src/.mesonpy-az8zmviv/meson-python-native-file.ini
      The Meson build system
      Version: 1.6.0
      Source dir:
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src
      Build dir:
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src/.mesonpy-az8zmviv
      Build type: native build
      Project name: dipy
[... 32 lines cut ...]
      Compiler for C++ supports arguments -Wno-non-virtual-dtor: YES
      Compiler for C++ supports arguments -Wno-sign-compare: YES
      Compiler for C++ supports arguments -Wno-switch: YES
      Compiler for C++ supports arguments -Wno-terminate: NO
      Compiler for C++ supports arguments -Wno-unused-but-set-variable: YES
      Compiler for C++ supports arguments -Wno-unused-function: YES
      Compiler for C++ supports arguments -Wno-unused-local-typedefs: YES
      Compiler for C++ supports arguments -Wno-unused-variable: YES
      Compiler for C++ supports arguments -Wno-int-in-bool-context: YES
      WARNING: OpenMP support not available because: Couldn't find libomp
      Run-time dependency OpenMP found: NO (tried system)
      Program brew found: YES (/opt/homebrew/bin/brew)

      ../dipy/meson.build:146:13: ERROR: Command `/opt/homebrew/bin/brew list libomp` failed with
      status 1.

      A full log can be found at
      /Users/richardhochenberger/Library/Caches/uv/sdists-v6/pypi/dipy/1.10.0/Q49o8JHK5MY3kcjcS8jYt/src/.mesonpy-az8zmviv/meson-logs/meson-log.txt

  help: `dipy` (v1.10.0) was included because `mne[dev]` (v1.8.0) depends on `mne-gui-addons`
        (v0.1) which depends on `dipy`
```
Similar situation: no binary wheel exists due to missing Python 3.13 support, compilation fails, hard stop.

I **still** believe we could provide a better user experience by adding an upper bound to our `pyproject.toml` anyway. I don't want to argue that this is "right" – there are good arguments for saying, we shouldn't have to worry about this. But I'm trying to put myself into a naïve user's shoes, who'll get flooded by the rather cryptic error messages above, which I believe are confusing.

drammock, 2024-12-14:
I think that's because Eric fixed the openmeeg problem yesterday, not because of an update to uv.

I know this isn't the point of the thread (else I would have mentioned it right away) but FYI it works to install dipy main and numba --pre. In other words both packages have fixes in the works that aren't released as stable yet.

hoechenberger, 2024-12-14:
Just to add:
> FWIW I also tried an approach relying on binary wheels exclusively:

the resolution I get here is the same I got when using `uv` 0.5.8 **without** the `--no-build` switch

cbrnr, 2024-12-14:
I think we should probably be more transparent which Python versions we support. We can say that core MNE supports 3.13, but some extra dependencies might not (yet) work. It would be even more helpful to specify the version that we know works with all extras though!

hoechenberger, 2024-12-15:
> I think that's because Eric fixed the openmeeg problem yesterday, not because of an update to uv.

You're right, even enabling the "old" resolution strategy now leads to the `numba` compilation error:
```shell
❯ uv tool install --python 3.13 --fork-strategy=fewest "mne[full]"
Ignoring existing environment for `mne`: the requested Python interpreter does not match the environment interpreter
Resolved 184 packages in 304ms
  × Failed to download and build `llvmlite==0.43.0`
  ╰─▶ Build backend failed to build wheel through `build_wheel` (exit status:
      1)

      [stdout]
      running bdist_wheel
      /Users/hoechenberger/.cache/uv/builds-v0/.tmp0NqWHs/bin/python
      /Users/hoechenberger/.cache/uv/sdists-v6/pypi/llvmlite/0.43.0/f0XXkh4tDXV8VrXnXRsJ0/src/ffi/build.py
      LLVM version...

      [stderr]
      Traceback (most recent call last):
        File
      "/Users/hoechenberger/.cache/uv/sdists-v6/pypi/llvmlite/0.43.0/f0XXkh4tDXV8VrXnXRsJ0/src/ffi/build.py",
      line 235, in <module>
          main()
          ~~~~^^
        File
      "/Users/hoechenberger/.cache/uv/sdists-v6/pypi/llvmlite/0.43.0/f0XXkh4tDXV8VrXnXRsJ0/src/ffi/build.py",
      line 229, in main
          main_posix('osx', '.dylib')
          ~~~~~~~~~~^^^^^^^^^^^^^^^^^
        File
      "/Users/hoechenberger/.cache/uv/sdists-v6/pypi/llvmlite/0.43.0/f0XXkh4tDXV8VrXnXRsJ0/src/ffi/build.py",
      line 142, in main_posix
          raise RuntimeError(msg) from None
      RuntimeError: Could not find a `llvm-config` binary. There
      are a number of reasons this could occur, please see:
      https://llvmlite.readthedocs.io/en/latest/admin-guide/install.html#using-pip
      for help.
      error: command
      '/Users/hoechenberger/.cache/uv/builds-v0/.tmp0NqWHs/bin/python' failed
      with exit code 1

  help: `llvmlite` (v0.43.0) was included because `mne[full]` (v1.8.0)
        depends on `numba` (v0.60.0) which depends on `llvmlite`
```

hoechenberger, 2024-12-16:
@larsoner What exactly did you change in OpenMEEG? :)

drammock, 2024-12-16:
> @larsoner What exactly did you change in OpenMEEG? :)

Look at the filenames on pypi. Latest release has `-abi-` in the filename and the wheels are marked as 3.10+. I don't fully get the details of how the building/packaging changed under the hood, but now when uv checks pypi for openmeeg it finds a wheel that is marked as 3.13 compatible. You can view the issue I linked above, and the PR that fixed it.

hoechenberger, 2024-12-16:
Thanks for the explanation, @drammock!!

hoechenberger, 2024-12-16:
So I suppose I can close this issue for now?

@cbrnr?

cbrnr, 2024-12-16:
Sure, unless we want to document our compatibility better (e.g., saying that we only support core MNE)? Actually, I don't even know how we handle that, so even if we don't add anything I'd be interested!

closed 2024-12-16 by hoechenberger