mne-tools/mne-python #13027: Selecting epochs from `EpochsTFRArray` objects raises obscure error due to incomplete `drop_log` initialisation

tsbinns, 2024-12-13:
### Description of the problem

If you create an `EpochsTFRArray` object with the default `drop_log=None` and then try to select epochs from the object, you get the following error:
[image]

This is because the `drop_log` is initialised as an empty `tuple`, so there is nothing for `_getitem()` to index:
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/time_frequency/tfr.py#L3108

<br>

This behaviour is in contrast to e.g. `EpochsArray` where `drop_log=None` initialises is as a tuple of empty tuples according to the number of epochs/events (which plays nicely with `_getitem()`):
https://github.com/mne-tools/mne-python/blob/b329515933915fd077495ea41de876119ac04c97/mne/epochs.py#L514-L518

<br>

Working around this is easy (e.g., `drop_log=tuple([() for _ in range(n_epochs)])`), but as a user I would expect the default `drop_log=None` to already work like this.

### Steps to reproduce

```Python
import numpy as np
from mne import create_info
from mne.time_frequency import EpochsTFRArray

n_epochs, n_chans, n_freqs, n_times = (5, 3, 20, 50)
sfreq = 100
data = np.random.rand(n_epochs, n_chans, n_freqs, n_times)
times = np.linspace(0, sfreq/n_times)
freqs = np.arange(1, n_freqs + 1)
info = mne.create_info(n_chans, sfreq, "eeg")
tfr = EpochsTFRArray(info=info, data=data, times=times, freqs=freqs, drop_log=None)
tfr[0]
```

### Expected results

`tfr[0]` should give the TFR data for the first epoch.

### Actual results

[image]

### Additional information

Using dev version.

drammock, 2024-12-13:
> in contrast to e.g. `EpochsArray` where `drop_log=None` initialises it as a tuple of empty tuples according to the number of epochs/events (which plays nicely with `_getitem()`)

seems like a fairly straightforward fix is to copy the `EpochsArray` behavior. Do you want to do a PR? I might get to it late next week, if you're busy.

tsbinns, 2024-12-13:
I'm happy to write it!

PR #13028 (2024-12-21): Fix `EpochsTFRArray` default drop log initialisation
#### Reference issue (if any)

Fixes #13027

#### What does this implement/fix?

Expands the logic for `EpochsTFRArray` default `drop_log` initialisation so `__getitem__()` works with default params.
closed 2024-12-21 by larsoner