mne-tools/mne-python #13033: Loading data file in CDT format

SueJaH, 2024-12-17:
### Describe the new feature or enhancement

Our university recently purchased MEG equipment from Neuroscan, which outputs data files in CDT format, landmark information in POM format, and other data in CEO, DPO, and HPI formats. We attempted to use the read_raw_curry function, but unfortunately, it did not work with these files. 

We were wondering if there is an updated function or feature in MNE that can handle these formats. We believe this addition would benefit many researchers, including us.

### Describe your proposed implementation

We expect a function that can load these formats.

### Describe possible alternatives

We have no idea with any alternative implementations.

larsoner, 2024-12-17:
Could you record some tiny dataset (maybe 1s?) and upload it somewhere? Then hopefully someone could have a look at improving `read_raw_curry` to work with your data

drammock, 2024-12-17:
xrefs:
- #12855 
- #12795 

In particular, we have permission (in https://github.com/mne-tools/mne-python/issues/12855#issue-2528909321) to basically copy / adapt the curry reader script provided by Compumedics. Can you confirm that your data are read successfully by https://github.com/neuroscan/curry-python-reader?  If so, then if you and/or your colleagues are game to try that adaptation @SueJaH, please feel free to open a PR!  Otherwise we will eventually get to it.

dominikwelke, 2025-03-21:
if @SueJaH is not interested i might tackle that PR @drammock . 
currently running into problems with curry9 cdt files, too

drammock, 2025-03-21:
> if [@SueJaH](https://github.com/SueJaH) is not interested i might tackle that PR [@drammock](https://github.com/drammock) . currently running into problems with curry9 cdt files, too

that would be a welcome contribution!