mne-tools/mne-python #13039: Using n_overlap in raw.compute_psd() fails if good data segments are shorter than n_overlap

moritz-gerster, 2024-12-19:
### Description of the problem

Using `n_overlap` only works if there are no good data segments shorter than `n_overlap`:
```
n_fft = int(raw.info["sfreq"])
spectrum = raw.compute_psd(method="welch", n_fft=n_fft, n_overlap=n_fft // 2)
```
This is clearly a bug as using 1s windows for Welch with 500 ms overlap is a very typical analysis. But it will fail if two bad segments are less than 500 ms spaced apart.

Importantly, this was no issue with my data in the past. The bug must have been introduced with some mne update (unfortunately, I don't know which one). 

### Steps to reproduce

```Python
import mne
from mne.time_frequency import psd_array_welch
import os

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample', 'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

annotations = mne.Annotations(onset=[5, 6.5], duration=[1, 1], description=['bad', 'bad'])
raw.set_annotations(annotations)

n_fft = int(raw.info["sfreq"])
kwargs = dict(n_fft=n_fft, n_overlap=n_fft // 2)

# psd_array_welch works fine
psds, freqs = psd_array_welch(raw.get_data(), n_fft, **kwargs)

# raw.compute_psd fails
spectrum = raw.compute_psd(method="welch", **kwargs)

# specifying n_per_seg explicitly does not help
assert kwargs["n_overlap"] < n_fft  # True
spectrum = raw.compute_psd(method="welch", **kwargs, n_per_seg=n_fft)

>> ValueError: noverlap must be less than nperseg.
```

### Link to data

[Jupyter Notebook Example](https://www.dropbox.com/scl/fi/ilyznafd28tpc1vcm04mh/mne_bug_spectrum_welch.html?rlkey=5rn58y6mv0bm2tpo87av957r1&dl=0)

### Expected results

The bad segments should be set to `np.nan` and good data segment windows with less than `n_fft` sample points should be discarded. This works with `n_overlap=None` but it does not work with `n_overlap > 0`.

### Actual results

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[7], [line 1](vscode-notebook-cell:?execution_count=7&line=1)
----> [1](vscode-notebook-cell:?execution_count=7&line=1) spectrum = raw.compute_psd(method="welch", **kwargs)

File <decorator-gen-336>:12, in compute_psd(self, method, fmin, fmax, tmin, tmax, picks, exclude, proj, remove_dc, reject_by_annotation, n_jobs, verbose, **method_kw)

File ~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2232, in BaseRaw.compute_psd(self, method, fmin, fmax, tmin, tmax, picks, exclude, proj, remove_dc, reject_by_annotation, n_jobs, verbose, **method_kw)
   [2229](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2229) method = _validate_method(method, type(self).__name__)
   [2230](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2230) self._set_legacy_nfft_default(tmin, tmax, method, method_kw)
-> [2232](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2232) return Spectrum(
   [2233](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2233)     self,
   [2234](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2234)     method=method,
   [2235](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2235)     fmin=fmin,
   [2236](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2236)     fmax=fmax,
   [2237](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2237)     tmin=tmin,
   [2238](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2238)     tmax=tmax,
   [2239](https://file+.vscode-resource.vscode-cdn.net/Users/moritzgerster/Library/CloudStorage/Dropbox/Code/BIDS_STN_LFP/notebooks/~/anaconda3/envs/local_sns013/lib/python3.12/site-packages/mne/io/base.py:2239)     picks=picks,
[... 104 lines cut ...]
Core
├☒ mne               1.7.1 (outdated, release 1.9.0 is available!)
├☑ numpy             2.0.1 (OpenBLAS 0.3.27 with 10 threads)
├☑ scipy             1.14.0
└☑ matplotlib        3.9.1 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.5.1
├☑ numba             0.60.0
├☑ nibabel           5.2.1
├☑ nilearn           0.10.4
├☑ pandas            2.2.2
├☑ h5io              0.2.4
├☑ h5py              3.11.0
└☐ unavailable       dipy, openmeeg, cupy

Visualization (optional)
└☐ unavailable       pyvista, pyvistaqt, vtk, qtpy, ipympl, pyqtgraph, mne-qt-browser, ipywidgets, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
├☑ mne-bids          0.15.0
├☑ eeglabio          0.0.2-4
├☑ edfio             0.4.3
├☑ pybv              0.7.5
└☐ unavailable       mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, mffpy

moritz-gerster, 2024-12-21:
I could now trace back where this error came from. The bug is absent in mne=1.6 and it exists from 1.7 on. 

I believe it was introduced with this bug fix #12536.

drammock, 2024-12-21:
> @drammock, is it possible that it was introduced with bug fixes #12536 or #12538? I checked the files but could not directly understand how the introduced changes impact overlapping windows.

Seems quite possible. I'll try to take a look next week

moritz-gerster, 2024-12-27:
Thanks! Please let me know if I can help @drammock, I would be more than happy to contribute :)

mistraube, 2025-10-30:
I did a `git bisect` using the script provided under `Steps to reproduce` and it confirmed that #12536 introduced the issue.

mistraube, 2025-10-31:
> [@mistraube](https://github.com/mistraube) are you up for trying to fix it? You seem to be on a roll this week :)

Yes, I looked into it and I think the problem is that SciPy sets `nperseq` to the `span length` if `nperseq` is greater than the `span length` https://github.com/scipy/scipy/blob/b1296b9b4393e251511fe8fdd3e58c22a1124899/scipy/signal/_spectral_py.py#L2425C1-L2429C35)

If we are in such a case and `noverlap` is greater than the `span length` we get the error.

So we need to handle these cases before calling SciPy's spectrogram method, but I am not sure what's the best way to do it.
I can think of:

1. Discard such small good spans completely
2. Set noverlap to some value smaller than the span length (in this case: what is a reasonable value?)

What do you think @drammock?

closed 2026-10-06 by larsoner