mne-tools/mne-python #13057: ICA get_sources() can't deal with a concatenated raw object

BeiGeJin, 2025-01-13:
### Description of the problem

I think there might be a bug in the _source_as_raw function, [here](https://github.com/mne-tools/mne-python/blob/maint/1.9/mne/preprocessing/ica.py#L1299) it assigns [self.last_samp] to self._last_samps, however self.last_samp is a property function and it seems that if its wrapped in a list, it won't actually call this function until used later.

### Steps to reproduce

```Python
import mne

# load data
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = sample_data_folder / "MEG" / "sample" / "sample_audvis_filt-0-40_raw.fif"
raw = mne.io.read_raw_fif(sample_data_raw_file).crop(0, 60).load_data()  # raw has 60 seconds of data
# concatenate
raw_concat = mne.concatenate_raws([raw.copy(), raw.copy()])  # raw_concat has 120 seconds of data
# do ICA
ica = mne.preprocessing.ICA(n_components=20)
ica.fit(raw_concat)
# get sources
raw_sources = ica.get_sources(raw_concat)  # but this will only have 60 seconds of data
print(raw_concat)
print(raw_sources)
```

### Expected results

raw_concat and raw_sources should have the same length of data.

### Actual results

But raw_sources only have the length of data corresponding to the first object in the list that we concatenate.

### Additional information

Platform             Linux-3.10.0-957.1.3.el7.x86_64-x86_64-with-glibc2.17
Python               3.12.2 | packaged by conda-forge | (main, Feb 16 2024, 20:50:58) [GCC 12.3.0]
Executable           [/home/jerryjin/miniconda3/envs/meg/bin/python](https://vscode-remote+ssh-002dremote-002bmind-002d0-002d12.vscode-resource.vscode-cdn.net/home/jerryjin/miniconda3/envs/meg/bin/python)
CPU                  x86_64 (16 cores)
Memory               46.9 GB

Core
├☒ mne               1.8.0 (outdated, release 1.9.0 is available!)
├☑ numpy             1.26.4 (MKL 2023.2-Product with 16 threads)
├☑ scipy             1.14.1
└☑ matplotlib        3.9.2 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.5.1
├☑ numba             0.60.0
├☑ pandas            2.2.2
├☑ h5py              3.11.0
└☐ unavailable       nibabel, nilearn, dipy, openmeeg, cupy, h5io

Visualization (optional)
├☑ qtpy              2.4.1 (None=None)
├☑ pyqtgraph         0.13.7
├☑ mne-qt-browser    0.6.3
├☑ ipywidgets        8.1.5
└☐ unavailable       pyvista, pyvistaqt, vtk, ipympl, trame_client, trame_server, trame_vtk, trame_vuetify

Ecosystem (optional)
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-icalabel, mne-bids-pipeline, neo, eeglabio, edfio, mffpy, pybv

BeiGeJin, 2025-01-13:
As a workaround, I change the code
```
out._first_samps = [out.first_samp]
out._last_samps = [out.last_samp]
```
to below
```
a = out.first_samp
b = out.last_samp
out._first_samps = [a]
out._last_samps = [b]
```
And it seems to solve my problem.

larsoner, 2025-01-16:
@BeiGeJin would you be up for making a quick PR to fix the issue? I would start by using TDD (test-driven development) to add a minimal new test to `mne/preprocessing/tests/test_ica.py` that causes the error you see at the top, then add your fix and see that the test passes.

closed 2025-11-05 by larsoner