mne-tools/mne-python #13066: despite adding events resample function returns only raw when sfreq already at desired frequency

Randomidous, 2025-01-15:
### Description of the problem

The[ resample function](https://github.com/mne-tools/mne-python/blob/maint/1.9/mne/io/base.py#L1387-L1390) takes an additional parameter "events" and returns a resampled copy if provided. This ensures that event marker onsets as resampled jointly with the data. However, when the sampling frequency of the input data is already at the desired frequency, the code exits only returning "raw":
```
sfreq = float(sfreq)
o_sfreq = float(self.info["sfreq"])
if _check_resamp_noop(sfreq, o_sfreq):
    return self
```

I think there should be an additional if statement that returns events as well if they were provided. Otherwise a ValueError can easily happen for streamlined preprocessing pipelines:
```
sfreq = float(sfreq)
o_sfreq = float(self.info["sfreq"])
if _check_resamp_noop(sfreq, o_sfreq):
    if events:
        return self, events
    else:
        return self
```

### Steps to reproduce

```Python
from mne.datasets import sample
from mne.io import read_raw_fif
import mne

data_path = sample.data_path()
raw_fname = data_path / 'MEG/sample/sample_audvis_raw.fif'
raw = read_raw_fif(raw_fname, preload=True)

# pretend dataset is recorded at 250 Hz
raw.resample(250)

# get events
events, event_id = mne.events_from_annotations(raw)

# now resample to desired freq
raw, events = raw.resample(sfreq=250, events=events)
```

### Link to data

https://mne.tools/stable/documentation/datasets.html

### Expected results

Opening raw data file [...\mne_data\MNE-sample-data\MEG\sample\sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Reading 0 ... 166799  =      0.000 ...   277.714 secs...
320 events found on stim channel STI 014
Event IDs: [ 1  2  3  4  5 32]
320 events found on stim channel STI 014
Event IDs: [ 1  2  3  4  5 32]

### Actual results

Opening raw data file [...\mne_data\MNE-sample-data\MEG\sample\sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Reading 0 ... 166799  =      0.000 ...   277.714 secs...
320 events found on stim channel STI 014
Event IDs: [ 1  2  3  4  5 32]
320 events found on stim channel STI 014
Event IDs: [ 1  2  3  4  5 32]
Sampling frequency of the instance is already 250.0, returning unmodified.

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[9], [line 16](...)
     [13](...) events, event_id = mne.events_from_annotations(raw)
     [15](.) # now resample to desired freq..
---> [16](...) raw, events = raw.resample(sfreq=250, events=events)

**ValueError: too many values to unpack (expected 2)**

### Additional information

Platform             Windows-11-10.0.22631-SP0
Python               3.12.7 | packaged by conda-forge | (main, Oct  4 2024, 15:47:54) [MSC v.1941 64 bit (AMD64)]
Executable           [.../python.exe](.../python.exe)
CPU                  Intel64 Family 6 Model 186 Stepping 3, GenuineIntel (12 cores)
Memory               15.7 GB

Core
├☒ mne               1.8.0 (outdated, release 1.9.0 is available!)
├☑ numpy             1.26.4 (OpenBLAS 0.3.23.dev with 12 threads)
├☑ scipy             1.11.4
└☑ matplotlib        3.8.1 (backend=module://matplotlib_inline.backend_inline)

Numerical (optional)
├☑ sklearn           1.3.2
├☑ numba             0.60.0
├☑ nibabel           5.3.2
├☑ nilearn           0.10.4
├☑ dipy              1.9.0
├☑ openmeeg          2.5.12
├☑ pandas            2.2.3
├☑ h5io              0.2.4
├☑ h5py              3.11.0
└☐ unavailable       cupy

Visualization (optional)
├☑ pyvista           0.44.1 (OpenGL 4.5.0 - Build 32.0.101.6078 via Intel(R) Iris(R) Xe Graphics)
├☑ pyvistaqt         0.11.1
├☑ vtk               9.3.1
├☑ qtpy              2.4.1 (PySide6=6.6.3)
├☑ ipympl            0.9.4
├☑ pyqtgraph         0.13.7
├☑ mne-qt-browser    0.6.3
├☑ ipywidgets        8.1.5
├☑ trame_client      3.4.0
├☑ trame_server      3.2.3
├☑ trame_vtk         2.8.11
└☑ trame_vuetify     2.7.1

Ecosystem (optional)
├☑ mne-icalabel      0.7.0
├☑ neo               0.13.4
├☑ eeglabio          0.0.2-4
├☑ edfio             0.4.4
2025-01-15 10:42:25,461 - XMLType - INFO - type <class 'mffpy.xml_files.XML'> cannot be registered
├☑ mffpy             0.10.0
├☑ pybv              0.7.5
└☐ unavailable       mne-bids, mne-nirs, mne-features, mne-connectivity, mne-bids-pipeline

sappelhoff, 2025-01-15:
Thanks for the report @Randomidous, I agree that in this case the `resample` method should return the unchanged events, and not just the unchanged raw.

larsoner, 2025-01-16:
@Randomidous would you be up for a quick PR to fix this behavior?

Randomidous, 2025-01-17:
@larsoner Find the pull request here: https://github.com/mne-tools/mne-python/pull/13070

PR #13070 (2025-01-24): BUGFIX: return events if provided when current = desired sfreq
#### Reference issue (if any)
[Fixes #13066](https://github.com/mne-tools/mne-python/issues/13066#issuecomment-2596286302)

#### What does this implement/fix?
The[ resample function](https://github.com/mne-tools/mne-python/blob/maint/1.9/mne/io/base.py#L1387-L1390) takes an additional parameter "events" and returns a resampled copy if provided. When the current is equal to the desired sfreq, only self is returned. This fix now returns raw and events if both were provided, avoiding preprocessing pipeline bugs down the line. 

#### A [... cut]
closed 2025-01-27 by Randomidous