mne-tools/mne-python #13095: Exporting to BrainVision creates extra marker

cbrnr, 2025-02-03:
Currently, exporting to BrainVision creates an extra marker containing the `meas_date`. Usually, this datetime is stored as the first marker, which looks like this:

```
Mk1=New Segment,,1,1,0,20231207111507000000
```

Importing that file into MNE again therefore implies that there is an additional marker that was not present before. In addition, I think the reader parses the measurement date so that this marker is not necessary at all.

I'm not quite sure how to handle this situation, so I'd appreciate your input @sappelhoff @larsoner @drammock. I came up with the following options:

1. Always ignore the first marker (if it is a `New Segment`).
2. Always ignore all markers of type `New Segment`.
3. Provide an option to ignore all `New Segment` markers.

I think a transparent solution without an additional parameter would be nice, WDYT?

sappelhoff, 2025-02-03:
The "New Segment" marker is not really an "experiment event", but rather information about the data. Continuous data will only have one of such markers (in the beginning) ... if you pause a recording in BV Recorder and then start it again, it will put another "New Segment" marker in at that point, as far as I know.

I don't really understand the problem yet, I think. Is it the following?

1. you have data with 3 markers A, B, C in some format
2. you import this data into MNE and then export it to BrainVision
3. the exported BrainVision data now has 4 markers: "New Segment", A, B, C
4. the "New Segment" marker is breaking pipelines that did not expect this marker

If this is the problem, then I don't think there is something to change, as the "New Segment" marker is sort of a normal BrainVision occurrence 🤔  if anything, I'd then opt for your proposal 3.

But happy to hear more about it and change my opinion.

cbrnr, 2025-02-03:
Yes, but for step 4 I would say:

4. When importing the BrainVision file, there are now four markers, so this breaks the round-trip.

If the first marker is a "New Segment", which only contains the recording date, we already parse this information and make it available in `info["meas_date"]`. There is no need to duplicate this information here (in an rather unexpected location). So I would rather go with my first option and drop the first marker if it is a "New Segment" (but of course provide the recording date in `info["meas_date"]`).

sappelhoff, 2025-02-03:
> When importing the BrainVision file, there are now four markers, so this breaks the round-trip.

Are you saying that when doing N operations of reading and then writing back to BV, we will add a new "New Segment" marker on each iteration? If yes, then I agree that this is off.

If however we write to BV for the first time and there is no "New Segment" marker, then I think we need to write it as the first marker, and put the meas date into it (where else would you put the meas date?).

cbrnr, 2025-02-03:
No, we're not adding a new marker on each iteration, but adding this one marker is still not right I think. Or rather, adding the marker is fine (because that's where you put the measurement date), but we should not import that marker as an annotation.

hoechenberger, 2025-02-03:
Maybe zu don't grasp the gravity of this issue fully, but I'd be +1 for simply ignoring this marker. I always have to manually drop this one after import, which is annoying

cbrnr, 2025-02-03:
Basically exactly what I'm proposing!

sappelhoff, 2025-02-03:
> Is the New Segment marker part of the file spec? I'm assuming yes, otherwise we wouldn't be adding it in the first place.

Just for completeness: Yes, Section 6.1 here: https://www.brainproducts.com/support-resources/brainvision-core-data-format-1-0/

[image]

And in the example in 6.3.2:

[image]

cbrnr, 2025-02-03:
Perfect, I'll prepare a PR then! Thanks for your input!

PR #13100 (2025-02-07): Skip first "New Segment" BrainVision marker
Fixes #13095. This PR skips the first "New Segment" BrainVision marker, as it only contains the recording time (which is already available in `info["meas_date"]`). I decided to not do anything with subsequent "New Segment" markers, as such changes could be unexpected.
closed 2025-02-07 by drammock