mne-tools/mne-python #13102: Add two additional changelog sections?

cbrnr, 2025-02-04:
Currently, all changes must fall into one of the [following categories](https://mne.tools/dev/development/contributing.html#describe-your-changes-in-the-changelog):

- notable
- dependency
- bugfix
- apichange
- newfeature
- other

I really miss the following two categories:

- changed (for any change which is not a bugfix, newfeature, or apichange)
- documentation (for changes to our documentation)

Could we add these two sections for more fine-grained categorization? For example, the change in #13100 would be "changed", but right now I had to put it into "bugfix", because it definitely is not any of the other categories.

hoechenberger, 2025-02-04:
I like that!

larsoner, 2025-02-04:
I think we previously discussed these possibilities, deciding documentation in https://github.com/mne-tools/mne-python/issues/12233#issuecomment-1858197294 and related discussions because it can be lumped in "other". For "changed" it is essentially the same as "other" already.

To me #13100 is a bugfix because it is a behavior change with no deprecation cycle. If it had been cosmetic or something it would have fit in "other"

drammock, 2025-02-04:
-1 from me, because:

- "changed" is equivalent to "other" (which we already have, and as @larsoner just noted)
- "documentation" changes are sometimes bugfixes (if the docs are incorrect) and sometimes "ehancements" (aka "newfeature") if they add a new tutorial or example. Other documentation changes I usually don't bother putting in the changelog, e.g., typo fixes, but they can go into "other" if they're more substantive or if folks want more visible credit.

cbrnr, 2025-02-04:
"other" is probably one of the worst possible names (together with "misc"), so since you say that it is equivalent with "changed", let's use that instead?

larsoner, 2025-02-04:
I think of "other" as a more narrow/precise term because it implies "any change that isn't caught by the other categories". The category "changed" is broader in that by its name it could encompass any change, and the reader has to infer more indirectly that it really means "other changes that don't fit in the above categories". In this sense "other" is clearer linguistically to me in its purpose than "misc" or "changes"

drammock, 2025-02-04:
> "other" is probably one of the worst possible names (together with "misc"), so since you say that it is equivalent with "changed", let's use that instead?

In the actual published changelog, it's "other changes". It's only "other" in the towncrier filename (`other.1234.rst`)

https://mne.tools/dev/changes/v1.9.html#other-changes

cbrnr, 2025-02-04:
Fair enough, but I don't think of it like that. It's a bit like "that change wasn't important to fit in any of the real categories, so let's put it in other". A "changed" category doesn't evoke this association. Same for a dedicated "documentation" category, it would highlight that doc changes are on the same level and not just "other".

cbrnr, 2025-02-04:
> In the actual published changelog, it's \"other changes\". It's only \"other\" in the towncrier filename (12345.other.rst)

That's perfect, I didn't know that!

closed 2025-02-04 by cbrnr