mne-tools/mne-python #13103: mne.filter.notch_filter default notch_widths too low?

SophieHerbst, 2025-02-04:
Hello,
when applying a notch filter to EEG data without setting the notch width (stop band width), the default is frequencies / 200. For typical 50 Hz line noise, this results in 0.25 Hz, which appears very narrow. The attached plot shows that the result is not satisfactory. 
I think it would be better to set a fixed band width, e.g. 1 Hz for all frequencies. 
What do you think?
Sophie

[image]

sappelhoff, 2025-03-04:
> It would probably be an annoying deprecation but probably worth it.

+1 to do this.

tsbinns, 2026-11-12:
Do we think this is something still worth tackling? I just ran into problems with this now and would be up for addressing. If we did, would it be the case of reducing the denominator in `freqs/x` or a fixed value like in the original post? I would lean towards the former but not set on anything.
CC @larsoner @drammock

drammock, 2026-11-16:
seems worth doing. I don't have a good sense of how broad these peaks are in the wild --- I know (from using UPSes that display input statistics) that the actual AC frequency can fluctuate, but IDK how much / what a sensible default notch BW would be. Maybe @BabaSanfour has insight here, or pointers to the relevant literature?  Absent that, I'm inclined to trust that the original author had a reason for making the BW a function of frequency, so I'd lean toward changing `1/200` to `1/100` or similar, rather than a fixed BW for all harmonics.

BabaSanfour, 2026-11-16:
I don't really have a specific number, but from my experiments I think it depends on the system. I ran into something similar while making ZapLine more robust in https://github.com/mne-tools/mne-denoise/pull/148: I limited ZapLine's cleaning to ±1 Hz around each harmonic. That caught almost all of the line on most recordings, but on a few clinical EEG recordings the line spread wider than that. So I think 0.25 Hz is too narrow.

Also, I test every ZapLine change on lab EEG, clinical EEG, CTF, MEGIN, 4D, KIT and ECoG data, so the data is ready. I could measure how wide the peaks really are and how much they drift, per system and per harmonic, so the new default is based on actual numbers rather than a guess. Happy to do it if that helps.

drammock, 2026-11-16:
> test every ZapLine change on lab EEG, clinical EEG, CTF, MEGIN, 4D, KIT and ECoG data, so the data is ready. I could measure how wide the peaks really are and how much they drift, per system and per [...]

yeah I think that would be useful info, if it's not much work to do!

BabaSanfour, 2026-11-16:
Working on it.

BabaSanfour, 2026-11-16:
The runs are done on ~1,300 recordings (lab EEG, clinical EEG, mobile EEG, iEEG, CTF, MEGIN, KIT and OPM). The line is narrow most of the time: 95% of its power is within ±0.1 Hz of 50/60 Hz, and it drifts by ~0.03–0.07 Hz. The current FIR default handles that really well. The issue is strength: strong lines have a wide skirt that a narrow notch leaves behind (probably what Sophie had; picture below). Separately, the drift grows with the harmonic (~2× at 2f, ~3× at 3f), so keeping the width proportional to frequency makes sense.

I also measured what each choice costs:
- `freqs / 100` helps for 30–50 dB lines without much distortion.
- `freqs / 50` helps a bit more but removes ~2–3× more signal at the harmonics.
- Very strong lines (> 50 dB) survive any width; those need IIR, spectrum interpolation (already implemented in mne-denoise), or a spatial method like ZapLine / ZapLine-plus.

Most of recordings (~1,200) come from one clinical EEG dataset: the default already removes the line in ~83% of the clinical EEG recordings and ~73% of the other ones; `freqs / 100` would bring that to ~87% / ~78% and `freqs / 50` to ~91% / ~87%. Given that small gain compared to the costs, I don't see a strong case for changing it: the special cases probably need proper methods anyway. If we do change it, I'd go with `freqs / 100`.

[image]

<details>
<summary>Claude Summary</summary>

**Data.** 1,299 recordings, raw (no prior filtering): 1,197 from one clinical EEG dataset (19 ch, 200 Hz, 60 Hz mains), and ~100 from public datasets: lab EEG (EEGBCI, LEMON, HBN/EGI, ERP CORE/BioSemi, MNE sample), mobile EEG, iEEG (OpenNeuro ds003029 from 4 US hospitals, ds003498 from Zurich, MNE epilepsy ECoG), CTF (Brainstorm, SPM faces, FieldTrip CMC), MEGIN (sample, somato, hf_sef, multimodal, MNE OPM-SQUID), 4D (HCP), KIT (MEG-MASC) and OPM (UCL, MNE OPM).

**Method.** 20 s Hann Welch spectra (a perfectly stable line measures ±0.1 Hz), floor fitted on both sides of each harmonic, widths only on channels where the line is ≥ 20 dB above the floor; nearby non-mains peaks (e.g. 59.08 Hz in MNE sample) excluded. Drift tracked in 10 s windows. Then `mne.filter.notch_filter` (FIR) with `f/200`, `f/100` and `f/50` on the 16 channels with the strongest line (300 s, filter edges excluded); the line counts as still visible if it stays > 3 dB above the original floor.

**How narrow is the line?** (median per recording, fundamental)

| system | 95 % of line power within | offset from 50/60 Hz | drift (p5–p95) |
|---|---|---|---|
| lab EEG | ±0.06 Hz | 0.01 Hz | 0.03 Hz |
| clinical EEG | ±0.08 Hz | 0.003 Hz | 0.06 Hz |
| iEEG | ±0.08 Hz | 0.02 Hz | 0.03 Hz |
| CTF | ±0.06 Hz | 0.004 Hz | 0.04 Hz |
| MEGIN | ±0.09 Hz | 0.04 Hz | 0.06 Hz |
| KIT | ±0.08 Hz | 0.02 Hz | 0.04 Hz |
| OPM | ±0.10 Hz | 0.03 Hz | 0.07 Hz |
| 4D | ±0.34 Hz (broad base) | 0.01 Hz | 0.03 Hz |

For reference, the default FIR notch attenuates ≥ 50 dB out to ±0.125 Hz (50 Hz) / ±0.15 Hz (60 Hz), then −26/−32 dB at ±0.2 Hz and almost nothing at ±0.5 Hz; `f/100` holds ≥ 50 dB out to ±0.25 / ±0.3 Hz.

**Harmonics.** Drift at 2f and 3f is 1.9–2.3× and 2.9–3.1× the fundamental's in every system with enough data, as expected from the mains frequency itself drifting. This comes from the public datasets only: at 200 Hz the clinical EEG has no harmonic below Nyquist.

**Line still visible after the notch** (% of recordings):

| | `f/200` | `f/100` | `f/50` |
|---|---|---|---|
| clinical EEG (1,073) | 17 % | 13 % | 9 % |
| other systems (98) | 27 % | 22 % | 13 % |
| of which 4D (12) | 83 % | 67 % | 0 % |
| of which mobile EEG (6) | 67 % | 33 % | 33 % |
| of which lab EEG (24) | 33 % | 33 % | 33 % |
| of which OPM (5) | 20 % | 20 % | 20 % |
| of which CTF / MEGIN / KIT / iEEG (51) | 0–12 % | 0–12 % | 0–7 % |

It depends mostly on line strength: below 30 dB the default removes the line in ~90 % of recordings (both clinical and other), wider notches help most for 30–50 dB lines, and above 50 dB about half survive any width (the skirt is wider than the notch, and the centre exceeds the FIR's ~55 dB stopband depth).

**Cost of a wider notch** (FIR, 4 harmonics, 1/f background):

| | spectrum removed at f / 4f | 30–100 Hz power lost | 100–250 Hz power lost | ringing after a step |
|---|---|---|---|---|
| `f/200` | 0.9 / 1.6 Hz | 1.3–1.9 % | 2.5–2.9 % | 1.0 % |
| `f/100` | 1.1 / 2.6 Hz | 1.7–2.5 % | 3.9–4.5 % | 1.5 % |
| `f/50` | 1.6 / 4.6 Hz | 2.5–3.8 % | 6.6–7.8 % | 2.6 % |

Ringing lasts about ±0.5 s for all three (zero-phase, so it also starts before the event).

**Caveats.** The clinical EEG is a single dataset, so the clinical rows reflect one site and system; the per-system rows are independent of it. Notch results are on the strongest channels of each recording, so they are a worst case. The cost numbers come from the filter responses, not from measuring lost brain signal in real data.

</details>