mne-tools/mne-python #13118: apply_function docstring wrong for epochs?

sappelhoff, 2025-02-18:
see: https://mne.tools/dev/generated/mne.Epochs.html#mne.Epochs.apply_function

[image]

☝ what is said here does not correspond to the actual data shape of epochs (3D arrays). So it is actually confusing how the passed function would deal with the epochs dimension.

See also:

https://github.com/mne-tools/mne-python/blob/64e01e54a3e149460c86a082ecb913993cd93dc6/mne/epochs.py#L1942

https://github.com/mne-tools/mne-python/blob/64e01e54a3e149460c86a082ecb913993cd93dc6/mne/utils/docs.py#L1863-L1871

Imama-Kainat, 2025-02-28:
Hi, I'd like to take on this issue. I've identified the incorrect docstring in mne/utils/docs.py and will submit a pull request with the corrected string. I'll also ensure to run the tests and build the documentation to verify the fix

shresth-keshari, 2025-02-28:
Well, I made the changes to the docstrings, but can't figure out how to test the changes by maybe viewing it in a locally hosted html file.
Please let me know if you have any suggestions.

drammock, 2025-02-28:
> I made the changes to the docstrings, but can't figure out how to test the changes by maybe viewing it in a locally hosted html file.

Instructions are in our contributor guide:

https://mne.tools/dev/development/contributing.html#building-the-documentation

Imama-Kainat, 2025-03-01:
> > I made the changes to the docstrings, but can't figure out how to test the changes by maybe viewing it in a locally hosted html file.
> 
> Instructions are in our contributor guide:
> 
> https://mne.tools/dev/development/contributing.html#building-the-documentation

Thanks for it

shresth-keshari, 2025-03-01:
> [@shresth-keshari](https://github.com/shresth-keshari) I noticed that we both worked on this issue. I've completed the changes, built the documentation, and submitted a pull request. Let me know if any adjustments are needed. Looking forward to feedback!

sure, let me know if I could help and thanks a lot !